OrcaSlicer/OrcaSlicer #10206: Slow printing down for better layer cooling is not working if printing by object. Auto cooling bug.

skarados, 2025-07-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Build 70931e5

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Ender 3 Pro with Klipper

### How to reproduce

1. Import multiple small parts on one plate
2. Enable printing sequence by object
3. Enable slow down printing for better layer cooling
4. Set the minimum speed when slowing down to a much lower value than your normal speed
5. Set the max fan speed threshold layer time to a higher value than the layer time for your small objects
6. Slice and confirm the layer time and printing speeds are not changing for the small objects

[image]

### Actual results

When you print by object and print multiple parts, the print is not slowing down.

[image]

If you only print that tiny object alone and do not change any setting, the print slows down normally.

[image]

### Expected results

The print should also slow down for better cooling, if you print objects with printing sequence "print by object".

### Project file & Debug log uploads

[keybak_idler_bracket.zip](https://github.com/user-attachments/files/21375847/keybak_idler_bracket.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

moth4017, 2025-07-22:
can you show the layer time of  the small part in both by object and plate as when i load your file the by object layer times to speed are correct

skarados, 2025-07-23:
Thanks for your reply. It might be confusing, since I have many parts in my file.. I began noticing it on plate number 4.

Layer times by object. The layer time of the highlighted part should be more than 5 sec, but is only 2.9sec (dark blue). The speed stays at 60mm/s but should be way less.
[image]

Layer times by plate
[image]

moth4017, 2025-07-23:
hi  you may get  a quicker response on a solution by going onto the orca slicer discord https://discord.gg/Jp39sAkn