OrcaSlicer/OrcaSlicer #10208: Parts count in plate group

tlhintoq, 2025-07-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I know its petty but it will be a nice helper to have the number of items on the plate show in the group.
`Clone - Fill` and suddenly you've got.... uhhhh... don't know... number of bolts on the plate.
When you need let's say...24 it would be nice to split that to common sense batches like 6 or 12 so you run even numbers of batches to get your final amount.  That sort of thing.

[image]

Just turning on labels doesn't really help when the parts are small and overlapping.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In the plates list, right after the plate name add  `(##)` for the number of parts on the plate.

[image]

### Describe alternatives you've considered

.

### Additional context

.

niklasb22, 2025-08-22:
Hi,
right-click on the desired plate and choose “Select all”.
In the bottom-right tooltip you’ll see the total number of selected objects.

[image]

tlhintoq, 2025-08-29:
> Hi, right-click on the desired plate and choose “Select all”. In the bottom-right tooltip you’ll see the total number of selected objects.

Good band aid if-until someone addresses it or closes it as 'not planned'.