OrcaSlicer/OrcaSlicer #10209: Is there any way that all filaments and print settings can JUST BE MADE to work with any printer?

christophervankammen, 2025-07-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

There are so many times where I just want to use the exact same print profiles and filament profiles for different printers. This is becoming an issue. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I think that there can simply be a way to overwrite the settings for Printer ID or simply just have a check box that allows me to use any print profile available. 

### Describe alternatives you've considered

I consider any thing that makes it possible to use any print profile or even have a print profile and filament profiles that are globally useable with out retrictions

christophervankammen, 2025-07-23:
I am open to undersatnding the code better if you have a general idea of where this code exists, then I could work on understanding what needs to be implemented for a template mode, a profile that is cross usable for all printers and / or filaments. that would be great. Otherwise, asking for help from more seasoned programmers of this repo would be why I put it here.

james633, 2025-07-23:
I think this is an unusual use case which would mislead the average new 3d printing users. It's not very many settings. Is it really so much trouble to type them in a few times? How many printers do you actually have?