OrcaSlicer/OrcaSlicer #10213: Negative brim object gap setting

elanderis, 2025-07-24:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

My part tends to pull out of the brim, negativing the usage of a brim entirely. I want some layer overlap between the brim and my first part layer.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The brim object distance setting could be set to allow for a negative value.

### Describe alternatives you've considered

1. I tried to make my own build of orcaslicer but havent been able to get pass an issue with CMake. I'm not tech savvy enough (even with the help of chatgpt.) 
2. Could be tied to my elephant foot compensation. 
3. Add a manual tapered brim in solidworks to my part. So there would be more than one layer holding onto the part.

discip, 2025-07-25:
@elanderis

> The brim object distance setting could be set to allow for a negative value.

No need for that, as ...

> 2\. Could be tied to my elephant foot compensation.

## ... this is the only culprit!

<br>

It was on my _'reporting list'_ for quite some time now, as I'm limited in that regard as well.

<br>

So the solution lies in making **`Brim-object gap`** depending on **`Elephant foot compensation`**.

@Noisyfox
Could you please have a look at this.

thanks in advance 😊

Noisyfox, 2025-07-25:
> making **`Brim-object gap`** depending on **`Elephant foot compensation`**

How exactly does this work?

discip, 2025-07-25:
Most likely I didn't express myself well. 😅

What I meant is:
That **`Brim-object gap`** should take **`Elephant foot compensation`** in to account, just like PrusaSlicer does.

[image]

[image]

[image]

e.g.:
If **`Brim-object gap`** is set to **`0`** and **`Elephant foot compensation`** has a **_value greater than_ `0`**, the brim should be attached directly to the perimeter of the first layer.

Hopefully that helps. 😊

elanderis, 2025-07-26:
I would like that to be fixed and then allow for me to genuinely have some overlap in the interface of the brim and the part. This would require brim object distance to have a negative value. @Noisyfox could this happen?

PhilBaz, 2025-08-19:
@Noisyfox Please see here and merge...this is one of the worst bugs that hasn't been addressed.

https://github.com/SoftFever/OrcaSlicer/issues/3720

closed 2026-02-17 by RF47