OrcaSlicer/OrcaSlicer #10227: anycubic kobra s1: "exclude object" not working

fazadaoc, 2025-07-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 70931e5

### Operating System (OS)

Windows

### OS Version

win 10

### Printer

anycubic kobra s1

### How to reproduce

enable _exclude object_, disable _label object_, slice and print

### Actual results

no option in the printer ui to skip any object. 

### Expected results

option to skip object. this happens as expected with the anycubic next slicer (based on orca)

### Project file & Debug log uploads

[Downloads.zip](https://github.com/user-attachments/files/21436017/Downloads.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

fazadaoc, 2025-07-25:
yep. it works fine with the anycubic next slicer (based on orca).... it even has an option in the slicer (see attachment)

The slicer also has other options such as timelapse (which i believe is in gcode but the option is not there in orcaslicer) as well as other proprietary options pre-print (see attachment). It'd be fantastic if those could be ported for anycubic but i'm unsure how feasable that might be

[image]
[image]

fazadaoc, 2025-07-26:
Top left the option to skip objects. First slices in anycubic orca, second in plain orca 

[image]

[image]

MxBrnr, 2025-07-26:
Looks like the difference is that in Anycibic-Orca you still have `Label Objects` turned on. Activate that in Orca along with Exclude and see if that makes a difference.

fazadaoc, 2025-07-26:
unfortunately that's not the case: i have tried with

- only _exclude_ and not _label object_ in ac orca (works)
- both options in anycubic orca (works)
- only _exclude_ in orca (does not work)
- both options in orca (does not work)

BUT weirdly, if i upload the file via normal orca then switch to the device tab in anycubic orca i can see the skip button! on the printer's display, however, there's no option to skip objects. i'm unsure why.

VigilantGardener, 2025-08-27:
I was able to get this working by altering the config for G-Code Thumbnails.  This solution was found by comparing GCode files generated by Orca and AC's Next Slicer.  Truly there were no significant differences.  The exclude object commands were available again on the printer screen after making this one change.

Details
AnyCubic Kobra S1 - No ACE
Firmware 2.5.6.0
Rinkhal's 7/25 release

I've been using this method to enable object exclude for months.  Preview image from Orca was taken from the 2.3.1-alpha release.

[image]