OrcaSlicer/OrcaSlicer #10228: Color Replacement

broncosis, 2025-07-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

it would be nice to be able to swap all of one color for another 
when you get pre colored models but your tool changer is loaded with different colors if you could say make everything that it tool 1 now tool 3 with out changing the machine would be amazing

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like a find and replace color option so that I don't have to change the order filaments are loaded in the machine 

IE replace filament one with filament two or tool 1 with tool 2 rather than trying to repaint the model which doesn't work that well

tlhintoq, 2025-07-26:
Can't you just can't the load in the `Filament` drop downs?
Paint is by slot number... so change your material selection.
Starts with 1 gray and 2 blue
[image]

[image]

Change that to 1 teal and 2 red

[image]

tlhintoq, 2025-07-28:
I still don't understand why you would repaint.
Just change the order in your slicer.

[image]

If paint color 1 is red, then change the dropdown for filament 1 to be whichever slot in your AMS is red.

Don't think of them as colors or even slots when you paint - its just color 1, color 2, color 3, color 4
Now in the materials section on the slicer match up the order you have them with the painting.
Color 1 can be slot 3 if that's the red
Color 2 can be slot 1 if that's the blue
...

tlhintoq, 2025-07-28:
I was starting to think I had forgotten something... So I had to double check.

You take a model and you paint it.  The number sequence when painting defaults to the assignments you set at the time.

[image]

Later if you want to change the coloring all you do is change the assignments in the slicer.  Or if they changed between when you painted it 6 months ago and your current load, just change the assignments.

[image]

broncosis, 2025-07-29:
ok bring a pre colored model in and then sync your ams colors and watch what happens to your model 
this is the case I have with my AMS 

I am also not always using a bambu AMS i am using a 5 head tool changer most of the time which is not as convenient to change  filaments and also doesn't have that drop down option 

[image]

broncosis, 2025-07-30:
@tlhintoq 
ok so I don't know why your fighting this feature request this is something that those of us in the tool changer community would like 
I understand its not a feature you need but a better way to map tools to parts of the model would be nice. 

ORCA does not sync with any printer that is not BAMBU  this is not a feature FOR BAMBU printers this would be a universal feature 
my device doesn't know what material is in it just that there is materiel present and it doesn't communicate with orca 

the drop down part  for the AMS works for that particular setup but for printing multi color and multi materiel print The ams is not the only way or even the best way. it is one way and unfortunately that part of the software was developed by bambu to work with Bambu hardware.   I am looking for a option that is not printer specific and will work with IDEX, bambu, AMS,  box turtle , and tool changing printers or any other multi color or material setup.

maybe a way to work around this would be to find a way to populate a list like the AMS has with the materials loaded in the machine 
manually.

SoftFever, 2025-08-01:
Coincidentally, I'm working on a similar feature (it should be, as it's too late at night for me now and I can only read the title). I should be able to submit a PR tomorrow.

broncosis, 2025-08-10:
tested it today from the nightly flatpack thanks seems to be working