OrcaSlicer/OrcaSlicer #10235: Make resolution value affect arachne resolution

Ataraxia-Mechanica, 2025-07-26:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

[image]
regardless of how small this value is set to, with arachne the walls always have a minimal resolution higher than this value

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Make these 3 values dynamically change with the resolution setting
[image]

valerii-bokhan, 2025-11-23:
Hello, I will try to look into it.

valerii-bokhan, 2026-02-23:
Oh, it is not fixed yet. It will be fixed in https://github.com/OrcaSlicer/OrcaSlicer/pull/11925

Can this issue be reopened, please?

closed 2026-04-26 by SoftFever