OrcaSlicer/OrcaSlicer #10244: Make it so that closing the window when changing printer settings reverts back to orignial printer settings

ckgdmm1, 2025-07-27:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When changing printer settings (eg. from 0.4 to 0.6 nozzle size) if I change my mind and close the pop up window, the printer setting is still changed and all the unsaved changes are lost. 
But this is not the case when changing filaments, when closing the window, it acts as if nothing happened, which is what it should be imo

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Make it so that when closing the pop up window when changing printer settings, it cancels the change

tlhintoq, 2025-07-28:
> But this is not the case when changing filaments, when closing the window, it acts as if nothing happened

Umm.... No.  If you make changes to a filament and close the window you still have those changes.  If you change the temp of a material and close the box, you're still working with the new temperature.  Its not *saved* as the permanent change, but it is the change for your project.  Basically this is how you make a temporary change that might be needed for a specific use-case but not a forever change to the material.

Start at 100% shrinkage:
[image]

Change to 97 then close without saving:
[image]

And the UI shows you the material has been changed but not saved.
[image]

ckgdmm1, 2025-07-29:
Sorry, I should've been more clear in my description. The system you guys are thinking about is indeed very good, my problem is a bit more niche

[image]

When you change any print settings, and you want to change filament, this window pops up asking if you want to transfer the change, discard the change or save the change and if you don't want anything to happen, you can close the window. This is perfect.

When you do the same for printer profile in the top left, you also get this window pop up, but if you close the window this time, deletes your setting and forces a printer profile change. I don't want this to happen, it should act as the same as the filament change.

Does this make sense?

ckgdmm1, 2025-07-30:
Ah yes, this is a problem more specific to my case. I have 2 profiles that I mainly use, one is the standard bambulab default, and one with G code that I modified to skip all the calibration steps and can start print in 30 secs. I only run the 2nd profile if I just finished printing something and I run the default when the printer's been idle or filament switch.

 In the future, if nozzle swapping becomes trivial, one may want to switch print profiles more frequently as well (between 0.2mm and 0.4mm and 0.6mm profiles for example)

But regardless, this is a simple fix I think, and it will make it more consistent with how the other window pop up works. Let me know what you guys think

tlhintoq, 2025-07-30:
Given the number of people that commented _"no, please don't change it for everyone because of one guy's niche condition"..._

Had you considered just having two printers defined?  One with the normal startup and one without? Then all you have to do is switch printer in the drop down.

- X1C calibrate
- X1C quickstart

[image]

Also - just curious - do you really run calibration every day?  I turned that sh!t off for good.  If I want to calibrate I'll send it on a true 20min calibration cycle.  Otherwise there's no reason to run it: The machine didn't suddenly change between Monday and Tuesday.
So just a thought - Just run the quickstart version unless you can see an actual reason to consciously make the choice to run a calibration cycle.

ckgdmm1, 2025-07-30:
I think I wasn't very clear with what i wanted in my post, so I don't think those people understood what I wanted, my change isn't very big, I just want to close the window and it not change printer. 

That is exactly what I am doing already, I have a set up for fast print and default print, like how you have one for silent50, if you change infill lets say and don't save it, and you change printer and you press cross, it forces the change, its just a small QoL

By calibration I just mean what the printer does automatically before each print, vibration, flow, etc. Takes like 5 mins, kind of annoying if you doing many iterations of a small project