OrcaSlicer/OrcaSlicer #10249: Travel without crossing wall to use smaller segment or G2/G3

Miranina, 2025-07-27:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

[image]

What's happening :

When slicing a circular object and "avoid crossing wall" on large movement in the infill it's using a low count of straight segment that trigger Klipper to use accell and decell. On newer machine where speed and accell are huge this create a large vibration and unnecessary stress on the mechanical parts. This also at the same time slow down the travel for no reason.

Tested with and without arc fitting. Tested as well with a standalone ArcWelder and in both case the segments of movement are left.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

What's expected : 

Either add a option to decrease the segment size of movement so klipper doesn't go into accell and decell ( as like during printing ) or make use of G2/G3 arc for these movement.

### Additional context

Anything else :

I've seen a similar issue on a bambu : https://youtu.be/ipf-sv_Zw3A?si=Rmhr6L18PkDiIhKD

My version :
https://cdn.discordapp.com/attachments/461153633631731712/1398859365800345752/20250726_224522.mp4?ex=68878d8e&is=68863c0e&hm=6a70bfb81842d98331781d634a14e3280a367e113275c53c9780d45b12931707&

MxBrnr, 2025-07-27:
G2/3 won't really work with Klipper because it just internally re-converts them back into straight lines anyway. All it does is add unnecessary and useless workload to the CPU/MCU.

Miranina, 2025-07-28:
True, Klipper convert them back into straight lines  but small enough and with shallow enough angles that it doesn't trigger the accell/decell that these large one do. I left the expectation it in the post as I'm unaware if G2/G3 would be easier to implement than shortening the straight lines and with a G2/G3 Klipper would convert it back into a smooth movement even if it's composed of hundreds of small lines.

Russellfam, 2026-11-05:
Is there any resolution to this yet? the issue has been open for a long time. I've got a similar issue, significant stuttering on the travel moves, the print moves are perfect