OrcaSlicer/OrcaSlicer #10260: Device Section of Orca Slicer not working with Flashforge AD5M

tbertw, 2025-07-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro

### Additional system information

The device option states "Connection Refused"

### Printer

Flashforge AD5M

### How to reproduce

1. Load item to Print
2. Edit
3. Select Slice
4. Send to printer to start and print
5. Go to Device
6. Look at refuse to connect message.

### Actual results

Hmmm… can't reach this page
192.168.1.97 refused to connect.
Try:

Checking the connection
[Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)
ERR_CONNECTION_REFUSED

### Expected results

Monitor process of printing

### Project file & Debug log uploads

Do not have that capability

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JRAltd, 2025-08-29:
Same Exact Issue.. Please Advise

melboone, 2025-09-21:
Any fixes? Same here

noravanq, 2025-10-07:
Same issue, both with 2.3.0 and 2.3.1.

I was able to send a print to the AD5M and print it from OrcaSlicer, but clicking on the Device tab shows "Could not connect to 192.168.29.221: Connection refused"

downloadblt, 2025-11-04:
Yea this sucks, the device tab worked fine when I first got the 5m pro, but after I got my p1s I think thats when it stopped working. not sure if its a coincidence but its really annoying. I'm getting closer and closer to trying forgex or one of the other klipper variants on my 5mpro. Since I factory reset the 5mpro I havent been able to sync flashprint so I need to get up and goto the printer every time.

melboone, 2025-11-05:
> Yea this sucks, the device tab worked fine when I first got the 5m pro, but after I got my p1s I think thats when it stopped working. not sure if its a coincidence but its really annoying. I'm [...]

This is a different issue. But to try and fix it, try with LAN only checkbox ticket. That's what worked for me to address that part.

Orca still doesn't connect

dmelsome-pi, 2026-11-13:
Same issue here with OrcaSlicer 2.5.0 on macOS with a Flashforge Adventurer 5M.

I've successfully configured the printer connection. The connection test passes, and I'm able to upload and print files over Wi-Fi without any issues.

The only thing I haven't been able to get working is the Device tab, which remains a blank white screen.