OrcaSlicer/OrcaSlicer #10280: Slicer not respecting 'top surface pattern' and only using one pattern

Altecheon, 2025-07-30:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Prusa Core One

### How to reproduce

1. Configuration
2. Flow Rate tests (1,2, and yolo)
3. go to strength, top/bottom shells
4. set a top surface pattern
5. slice
6. observe last layer doesn't change when updating the surface pattern

### Actual results

If printed you end up with circle patterns rather than the lines that should print if setting monotonic

### Expected results

Should see final layer be what is set in the config.

### Project file & Debug log uploads

[Orca YOLO Flow Calibration.zip](https://github.com/user-attachments/files/21517023/Orca.YOLO.Flow.Calibration.zip)

[debug_Wed_Jul_30_10_02_09_13468.log.zip](https://github.com/user-attachments/files/21517032/debug_Wed_Jul_30_10_02_09_13468.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

MxBrnr, 2025-07-31:
Not a bug. Top surface pattern is set per individual object here, so the global setting doesn't apply.

ianalexis, 2025-07-31:
> with new instructions on how to use it. [](https://github.com/SoftFever/OrcaSlicer/discussions/9609#discussioncomment-13395354)

Will update the wiki with that.

Altecheon, 2025-07-31:
Thank you for the prompt response. I apologize for still learning. Thank you for sharing the discussion on the new instructions.