OrcaSlicer/OrcaSlicer #10288: Cold wipe at start

aalencia, 2025-07-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Elegoo Centauri Carbon

### How to reproduce

After slice and print, G code doesn’t fully heat nozzle before the first wipe resulting in possible blobs.  

### Actual results

Blobs I say.  

### Expected results

Clean nuzzle.  We hate blobs.

### Project file & Debug log uploads

There is none.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

aalencia, 2025-08-01:
Also, I swapped the bed heat and nozzle heat so that the bed heats first.  it can take a long time. We don’t want the nozzle just sitting there dripping.

SoftFever, 2025-08-03:
Which machine model? These settings are machine specific.

aalencia, 2025-08-03:
.ELEGOO Centauri carbon.

closed 2025-11-09 by github-actions[bot]