OrcaSlicer/OrcaSlicer #10292: Multicolor Printig von Kobra 3 V2 - Always uses settings for bed level of Color T0

jolo1581, 2025-08-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1

### Printer

Anycubic Kobra 3 V2 Combo Firmwareversion 1.0.7.3) (Profile of Anycubic Kobra 3)  Rinkhals

### How to reproduce

ACE Pro:
- T0 = PLA
- T1 = PLA
- T2 = PLA
- T3 = PETG

When I only want to use T3 (PETG) and I slice my model and Print it, the Bed Leveling changes temperature of T0.

After Bed Leveling the print starts and it changes the temperature to the right nozzle and bed temperature.

I fixed the problem by changing T0 to PETG for slicing, but I think that should not the normal way.

Steps with failure:
1. Add STL
2. Change color to filament 4
3. Slice model
4. Upload and Print

Steps with good result:
1. Add STL
2. Change color to filament 4
3. Change material on filament 1 to same of filament 4
4. Slice model
5. Upload and Print

Maybe I'm to stupid to work right with Orca, but I think this might be a bug

### Actual results

bed heats up on settings of filament 1 and then uses temperature settings for filament 4 while printing

### Expected results

bed and printing temperature should be on selected material 

### Project file & Debug log uploads

[Orca.zip](https://github.com/user-attachments/files/21547048/Orca.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

DerDoubleD, 2025-08-29:
+1
same here

jolo1581, 2026-11-03:
Since Version 2.4.2 Orca slicer does 't work anymore with Kobra 3V2 so I can't confirm that this bug is solved