OrcaSlicer/OrcaSlicer #10293: I cannot get out of basic mode to adjust anything

Gary-Cohn, 2025-08-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

elegoo centauri carbon

### How to reproduce

I can download files no problem

### Actual results

I downloaded the latest version of orca slicer 2.3.0
I cannot get out of basic mode. there is no way to adjust any files.
I need advanced mode to start using the program

### Expected results

I have no problem loading files

### Project file & Debug log uploads

I have no problem loading files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MxBrnr, 2025-08-01:
Screenshots would be very helpful here.
Can try going into Preferences and turning on Developer Mode, it's basically the same thing.

Gary-Cohn, 2025-08-02:
I went into preferences and changed to developer mode. It fixed the problem.
Thank you.
It would be helpful if there was another way to get into advanced mode or
at least be able to change modes
I am a happy camper now.
Thanks again,
Gary

closed 2025-11-09 by github-actions[bot]