OrcaSlicer/OrcaSlicer #10334: Device View: Save zoom facotr issue number #2285

TheCorruptedEngineer, 2025-08-05:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Its been 2 years since someone else posted about this why is there still no way for the device window to remember the zoom factor
people have to use browser windows instead of using the actual device tab in the slicer because of how annoying this is. it shouldn't even be difficult to fix just add an option for the scale factor in settings for the device window

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

add an option for the scale factor in settings for the device window

thelegendtubaguy, 2026-11-08:
This is a dupe of #8889 and should be closed.

raistlin7447, 2026-11-08:
Thanks for the request. This asks for the same thing as #8889 (Device tab zoom not being remembered), so I'm closing this one as a duplicate to keep the discussion in one place. Please add a 👍 or any extra details there. If your request covers something #8889 doesn't, reply here and we'll reopen it.

closed 2026-11-08 by raistlin7447