OrcaSlicer/OrcaSlicer #10355: Calibration features gone

GETudury, 2025-08-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Labs X1C

### How to reproduce

Calibration features like calibration tower and others are gone

### Actual results

Calibration features like calibration tower and others are gone

### Expected results

Calibration features like calibration tower and others are gone

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MxBrnr, 2025-08-08:
Well this is pretty much of no help. Filling out the form as instructed usually works much better for getting issues resolved. Don't expect any traction on this in its current state.

coryrc, 2025-08-09:
Click Prepare.
Move your cursor over the File menu (don't click).
Move your cursor straight to the right and click Calibration. (This is NOT the same "Calibration" as in your screenshot).

IMO the Calibration you did click on should be removed, sometimes I click it and have a "wtf!?" moment until I remember the above.