OrcaSlicer/OrcaSlicer #10357: Tree supports start in mid air - Still

tlhintoq, 2025-08-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev build bb4dbd7

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

[image]

### Printer

All that I've tried, Giga, Neptune, Bambu

### How to reproduce

1. Use auto tree supports - Yeah, that about covers it.

[image]

### Actual results

Supports start in mid air - they look like the magically floating cliffs in the first Avatar movie.

### Expected results

Don't start supports in the middle of the air.

### Project file & Debug log uploads

[FloatingSupports.zip](https://github.com/user-attachments/files/21696342/FloatingSupports.zip)
Video attached

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Video attached

tlhintoq, 2025-08-12:
Another job... countless variations on the setting... trunk size... branch angle... nothing closes the holes in the tree trunks... nothing stops the supports from beginning in mid air.

**_"Just play around until you magically stumble over something that works"_** is not a strategy.

- Supports that start in mid air
- Support trunks that suddenly jump out 2mm bigger in diameter trying to make an unsupported support loop.

EvansJahja, 2026-11-11:
Unstale