OrcaSlicer/OrcaSlicer #10359: Incorrect filling of the first layer of the tree-like hybrid support causes problems with the support base.

Palagach, 2025-08-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Snapmaker J1s

### How to reproduce

1. Enable tree support in the process profile settings and select the support style: slim, strong, or hybrid.
2. Set the extrusion width for the first layer to 0.8mm and for the support layer to 0.4mm.
3. Turn on the preview and start printing.

### Actual results

When printing the first support layer, the preview shows that the first support layer has a lot of narrow fill lines instead of the standard 0.8 mm fill lines (such as the default or organic tree support). 

[image]

> The behavior of filling the first layer of tree support in the "slender", "strong", and "hybrid" styles. The width of the first layer is set to 0.8 mm, but in the screenshot above, this condition is not met except for one outer line of the first layer. There is no separate setting for adjusting the first layer of support in Orca.

[image]

> The behavior of filling the first layer of tree support with the "Default" and "Organic" styles. (which I consider to be completely correct and acceptable)

 This issue applies to the slim, sturdy, and hybrid styles, causing the nozzle to heat up the first support layer during printing because it has to print many 0.4 mm lines instead of 0.8 mm lines. This causes the first layer to accumulate a lot of plastic, and instead of being 0.2 mm thick, it grows to 1-1.5 mm. The nozzle may spread the uncooled plastic over the printing area, and the first layers of supports may be printed incorrectly. I can't set the extrusion width of the support and the first layer to be the same, as the 0.4 mm support is easier to detach without leaving any traces, and it is better than the 0.8 mm support. However, I need the first layer to be 0.8 mm to ensure excellent adhesion to the table.

[image]

[image]

[image]

[image]

### Expected results

When slicing the first layer of tree-shaped supports with the "Slim", "Strong", and "Hybrid" styles, Orka will make all the lines of the first layer 0.8mm wide, as specified in the extrusion width parameters, as it works for supports with the "Default" and "Organic" styles. Anything above the first layer should be printed according to the support width parameters, which in my case is 0.4mm. Either add a separate setting for the first support layer, or make the first layer equal to the extrusion width of the first layer for all lines in that layer.

### Project file & Debug log uploads

[support_test_low.zip](https://github.com/user-attachments/files/21698387/support_test_low.zip)

[Logfile.zip](https://github.com/user-attachments/files/21698394/Logfile.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

CodyReichert, 2025-09-16:
I'll add that I'm also experiencing this on my Ender 3 Pro printer profile, but _not_ on my K1C printer profile.

scaiper, 2025-09-17:
I'm having the same issue.

It appears that extrusion width is using the "First layer line width" option, but line spacing is incorrectly using the "Support line width". The result is that extrusion width do not match line spacing.

As a workaround I set "First layer line width" == "Support line width".