OrcaSlicer/OrcaSlicer #10381: Linux builds are broken at HEAD

coryrc, 2025-08-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

HEAD 90b59ff7d6

### Operating System (OS)

Linux

### OS Version

Debian sid/trixie

### Printer

N/A

### How to reproduce

`./build_linux.sh -j 2 -dsr`

Additionally, the more important RelWithDebInfo cannot be built using the scripts (or probably at all, TBD)

### Actual results

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Boost-1.83.0/BoostConfig.cmake:141 (find_package):
  Could not find a configuration file for package "boost_filesystem" that
  exactly matches requested version "1.83.0".

  The following configuration files were considered but not accepted:

    /opt/prg/OrcaSlicer/deps/build/destdir/usr/local/lib/cmake/boost_filesystem-1.84.0/boost_filesystem-config.cmake, version: 1.84.0

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.83.0/BoostConfig.cmake:262 (boost_find_component)
  CMakeLists.txt:490 (find_package)
```

I think it's finding my system boost library and not the deps one.

### Expected results

It should create the executable.

### Project file & Debug log uploads

[cmake-dsr-upstream-main.log](https://github.com/user-attachments/files/21724647/cmake-dsr-upstream-main.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

coryrc, 2025-08-12:
The only build I can get to work on Linux is the Debug build, but it throws asserts all over in use and is impossible to use without the debugger and many breakpoints to head off the many problems found.

coryrc, 2025-08-12:
```
/opt/prg/OrcaSlicer/deps/build/destdir/usr/local/lib/cmake/boost_filesystem-1.84.0$ rgrep 1.84
boost_filesystem-config.cmake:# Generated by BoostInstall.cmake for boost_filesystem-1.84.0
...
```

Not sure why.

coryrc, 2025-08-12:
Wait.... https://github.com/SoftFever/OrcaSlicer/blob/main/deps/Boost/Boost.cmake

It's explicitly installing 1.84.0

and

CMakeLists.txt is explicitly asking for 1.83.0 https://github.com/SoftFever/OrcaSlicer/blob/90b59ff7d65ac17ef37e46a5452c16671522fdc3/CMakeLists.txt#L490

Ugh, is this because I have a relatively new version of cmake? I've been getting lots of build warnings that say "such-and-such without an option is deprecated starting <recent version>"

coryrc, 2025-08-12:
```
$ cmake --version
cmake version 3.31.6

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

FTA:

The [version] argument requests a version with which the package found should be compatible. There are two possible forms in which it may be specified:

* A single version with the format major[.minor[.patch[.tweak]]], where each component is a numeric value.

* A version range with the format versionMin...[<]versionMax where versionMin and versionMax have the same format and constraints on components being integers as the single version. By default, both end points are included. By specifying <, the upper end point will be excluded. Version ranges are only supported with CMake 3.19 or later.

so... how is this building for anyone!?

coryrc, 2025-08-12:
From the [mostly-pushed commit build action for Ubuntu](https://github.com/SoftFever/OrcaSlicer/actions/runs/16885934537/job/47833630845):

```
-- Found Boost: /home/runner/work/OrcaSlicer/OrcaSlicer/deps/build/destdir/usr/local/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.83.0") found components: system filesystem thread log log_setup locale regex chrono atomic date_time iostreams program_options nowide 
```

and

```
cmake version:~3.28.0
```

Hmmm

coryrc, 2025-08-12:
FOUND IT

https://cmake.org/cmake/help/latest/policy/CMP0167.html

"This policy was introduced in CMake version 3.30. It may be set by [cmake_policy()](https://cmake.org/cmake/help/latest/command/cmake_policy.html#command:cmake_policy) or [cmake_minimum_required()](https://cmake.org/cmake/help/latest/command/cmake_minimum_required.html#command:cmake_minimum_required). If it is not set, CMake warns, and uses OLD behavior."

So because I have a newer version of cmake, it won't build, because I'm assuming the "upstream `BoostConfig.cmake`" matches exactly the version (as `find_package()` is supposed to) as opposed to old behaviour, which was incorrect but treated it as a minimum version.  #9527 @anarsoul

coryrc, 2025-08-12:
Nope, the build system seems to following the same path, I found `-- Using qhull from system.` in the logs.

coryrc, 2025-08-13:
I don't think related, but:

From PrusaSlicer/doc/Dependencies.md:

```
## External libraries in source tree
...
* qhull: libqhull-dev does not contain libqhullcpp => link errors. Until it is fixed, we will use the builtin version. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=925540
```

Problem is that library includes headers but not the static built library. Apparently "fixed" five years ago, following to the diff:

```
-add_library(${qhull_CPP} STATIC ${libqhullcpp_SOURCES})
+add_library(${qhull_CPP} SHARED ${libqhullcpp_SOURCES})
```

So just add_library isn't including the library, but we do have it as shown in previous comment.

coryrc, 2025-08-13:
Okay, so lets look at the QHull source CMakeLists.txt; how does it use `target_link_libraries`? Excerpting relevant:

```
    set(qhull_CPP qhullcpp)
    set(qhull_STATICR qhullstatic_r)
...
add_library(${qhull_CPP} STATIC ${libqhullcpp_SOURCES})
set_target_properties(${qhull_CPP} PROPERTIES
    VERSION ${qhull_VERSION}
    POSITION_INDEPENDENT_CODE "TRUE")
...
# Define user_eg3 linked to qhullstatic_r and qhullcpp static library
...
target_link_libraries(user_eg3 ${qhull_CPP} ${qhull_STATICR})
```

Aha! It's using `qhullcpp` and not `Qhull::qhullcpp`. Where does that come from?

https://github.com/prusa3d/PrusaSlicer/commit/6136fe7d92e386f1a6a6ab227f49c7d6a68abbb3

Aha! A completely vague "Future-proof qhull dependency handling" actually ends up making it more brittle. Quelle surprise

coryrc, 2025-08-16:
> > FOUND IT
> > [cmake.org/cmake/help/latest/policy/CMP0167.html](https://cmake.org/cmake/help/latest/policy/CMP0167.html)
> > "This policy was introduced in CMake version 3.30. It may be set by [cmake_policy()](https://cmake.org/cmake/help/latest/command/cmake_policy.html#command:cmake_policy) or [cmake_minimum_required()](https://cmake.org/cmake/help/latest/command/cmake_minimum_required.html#command:cmake_minimum_required). If it is not set, CMake warns, and uses OLD behavior."
> > So because I have a newer version of cmake, it won't build, because I'm assuming the "upstream `BoostConfig.cmake`" does not [claim compatibility with the requested version](https://cmake.org/cmake/help/latest/command/find_package.html#version-selection). [#9527](https://github.com/SoftFever/OrcaSlicer/pull/9527) [@anarsoul](https://github.com/anarsoul)
> 
> This shouldn't affect anything as the OLD behavior is used by default. Even then, this is only referring to using the config file generated by boost to find the library, which we already do.

But we already override the default: https://github.com/SoftFever/OrcaSlicer/blob/721ce0cb6d60fccfa1dc02d54f1c11b0fa342d16/CMakeLists.txt#L488

So prior to cmake 3.30, the policy does nothing.

> > Yup, the [upstream library uses the EXACT keyword when searching for Boost](https://github.com/boostorg/boost_install/blob/c9fbbd023650a85e80a649b1d5fb5c40c1fc15eb/BoostConfig.cmake#L141).
> 
> I don't believe this means much to our project. It looks like the reason that script is using exact mode is because it is looking for the rest of the boost components. In that script, `Boost_VERSION` would refer to the version of BoostConfig found, and since the components need to be the same version, the EXACT keyword is used.

I don't know if that's true, but changing the version to match fixed the problem.

> As for Qhull: "Using qhull from system" means that rather than building the in-source, modified version of qhull (`src/qhull/src`), it will link against a library found on the system. OrcaSlicer builds qhull as a part of the deps, so it should always use the system qhull library. Since you are getting errors from the parts of `src/qhull/CMakeLists.txt` that are trying to configure building the in-source version of qhull, I would suspect your system is also having trouble finding the deps built version of qhull.

I do not think this is correct. `src/qhull/CMakeLists.txt` is always used; for example, it's the only source of the string `"Using qhull from system."` which you'll find in CI build logs. There is not qhull under `cmake/modules` and https://github.com/SoftFever/OrcaSlicer/blob/721ce0cb6d60fccfa1dc02d54f1c11b0fa342d16/src/qhull/CMakeLists.txt#L32 exists when its not found in order to do a build. https://github.com/SoftFever/OrcaSlicer/blob/721ce0cb6d60fccfa1dc02d54f1c11b0fa342d16/src/CMakeLists.txt#L14 always includes the qhull directory.

> It seems like these issues may be specific to your machine/config. Are there any CMake environment variables set that could be causing issues? Are the correct permissions set on the OrcaSlicer source folder? Could you provide the full build output (it looks like the one provided was not done cleanly)? Does using PR [#10294](https://github.com/SoftFever/OrcaSlicer/pull/10294) solve this issue?

I was trying to use your PR but I was getting all these dumb errors! So I went back to HEAD and it was still broken. #10386 fixes them. After submitting my PR, now it's revealed that the macos-14 deps build is broken at HEAD, but it turns out that doesn't run automatically... so I filed #10405 and started working on #10404 so I can prove it's not my change.

I just want a RelWithDebInfo build on Linux!!!!

Ocraftyone, 2025-08-17:
> But we already override the default:

We already use the method the new policy specifies, even on older versions. The new policy related to boost is that you need to use the config system for finding the boost library rather than the old method of using the FindBoost.cmake module. This policy should not be an issue.

> I don't know if that's true, but changing the version to match fixed the problem.

I was able to test and confirm this by adding `-DBoost_DEBUG` to the cmake command when building the slicer. This enables a more verbose output for `BoostConfig.cmake` and will print out `Boost_VERSION`. The printed version corresponds to the library version that was found, not the one specified by the find_package command. As I said above, something is causing CMake to look for exact package versions, even when EXACT is not specified.

> I do not think this is correct. `src/qhull/CMakeLists.txt` is always used; for example, it's the only source of the string `"Using qhull from system."` which you'll find in CI build logs. There is not qhull under `cmake/modules` and
> 
> [OrcaSlicer/src/qhull/CMakeLists.txt](https://github.com/SoftFever/OrcaSlicer/blob/721ce0cb6d60fccfa1dc02d54f1c11b0fa342d16/src/qhull/CMakeLists.txt#L32)
> 
> Line 32 in [721ce0c](/SoftFever/OrcaSlicer/commit/721ce0cb6d60fccfa1dc02d54f1c11b0fa342d16)

The `else (Qhull_FOUND)` line is an else keyword for the above `if(Qhull_FOUND)` statment. In else statments (and others like endfunction, endif, ...) you can optionally provide the statement of the corresponding block start. So the block below that is if Qhull is NOT found. [CMake docs](https://cmake.org/cmake/help/latest/command/else.html)

If Qhull is found, that library is used. If not, then it builds the in-source version. OrcaSlicer always builds Qhull as a part of the deps (found in deps/Qhull) and should always be found. The code that builds the in-source Qhull is, at this time, dead code. If that is being run, something else is broken.

I will still try to do more testing in a true VM when I get a chance. It is possible there is something causing issues with Debian Trixie. Finding the root cause and fixing that will be better in the long term 😁.

coryrc, 2025-08-17:
> We already use the method the new policy specifies, even on older versions. The new policy related to boost is that you need to use the config system for finding the boost library rather than the [...]

Ah, you're saying in 3.28 it should already be using the new method. Hmm. Maybe the policy isn't taking effect? It is in an if statement I didn't understand [(though seems legit?)](https://cmake.org/cmake/help/latest/command/if.html#policy).

I don't know. I know I compiled everything successfully once. Then I've been compiling some other stuff, system things got upgraded, all of a sudden I could no longer compile Orca, even going back to the same commit. Took me a while to get around to seeing my cmake is much newer (I suppose I could prove it by downgrading cmake...).

> > I do not think this is correct. `src/qhull/CMakeLists.txt` is always used; for example, it's the only source of the string `"Using qhull from system."` which you'll find in CI build logs. ...
> The `else (Qhull_FOUND)` line is an else keyword for...

I was responding to:

> Since you are getting errors from the parts of src/qhull/CMakeLists.txt that are trying to configure building the in-source version of qhull

I am getting it from https://github.com/SoftFever/OrcaSlicer/blob/721ce0cb6d60fccfa1dc02d54f1c11b0fa342d16/src/qhull/CMakeLists.txt#L25 which is the portion for using the found library. I may have pasted the wrong link (there's a lot of links).

> I will still try to do more testing in a true VM when I get a chance. It is possible there is something causing issues with Debian Trixie. Finding the root cause and fixing that will be better in the long term 😁.

The qhull cmake never should have been changed from upstream. I don't know why it broke, but there wasn't any good reason to differ in the first place. My PR restores it to match the current upstream and it works for me and in the checks. I'm glad it isn't one of the problems you have hit!

Ocraftyone, 2025-08-21:
I can confirm that OrcaSlicer builds correctly on a fresh install of Debian Trixie in a VM

coryrc, 2025-08-22:
> I can confirm that OrcaSlicer builds correctly on a fresh install of Debian Trixie in a VM

Well damn, I'm at a loss why it doesn't work on mine. The changes I proposed are still good though. Worse case I can just apply them locally...