OrcaSlicer/OrcaSlicer #10382: `Top Interface Spacing` disabled for all all `Interface pattern` options.

tlhintoq, 2025-08-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev build 53440fd

### Operating System (OS)

Windows

### OS Version

11

### Printer

All, but noticed on giga

### How to reproduce

Turn on supports.  Normal (auto)/default in this example

### Actual results

`Top interface spacing` is disabled for all patterns.

[image]

### Expected results

Shouldn't be disabled.

### Project file & Debug log uploads

.

### Checklist of files to include

- [x] Log file
- [x] Project file

discip, 2025-08-12:
@tlhintoq
This is due to the fact that you are using **`Ironing support interface`**.
See the tooltip for reference. 😊

[image]

tlhintoq, 2025-08-12:
Ahhh... Okay.  Thank you!