OrcaSlicer/OrcaSlicer #10404: Build Deps isn't running in CI regularly

coryrc, 2025-08-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

HEAD

### Operating System (OS)

macOS

### OS Version

macos-14

### Printer

N/A

### How to reproduce

1. `Build Deps` is failing [on my PR](https://github.com/SoftFever/OrcaSlicer/pull/10386), even though it shouldn't. Go looking for most recent runs.
2. First went to https://github.com/SoftFever/OrcaSlicer/actions/workflows/build_all.yml and browsed through. Several dozen without having run `Build Deps`.
3. Filter to just push events: https://github.com/SoftFever/OrcaSlicer/actions/workflows/build_all.yml?query=event%3Apush
4. Even in just pushes, the last run which ran Build Deps was 3 days ago: https://github.com/SoftFever/OrcaSlicer/actions/runs/16871883036/job/47788101507

(BTW I actually looked at dozens of runs for PRs and pushes and put the links here, but the tab got unloaded and it was all deleted)

### Actual results

It's really hard to find a run, and it isn't that recent.

### Expected results

It should be easy to find a recent run on HEAD which has run all jobs.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

We should set up a CI workflow which runs all the build steps once a day and creates alerts/issues/something when it is failing.

coryrc, 2025-08-26:
/assign-me

coryrc, 2025-10-22:
Still waiting on review.

SoftFever, 2025-11-23:
Hi @coryrc 

I'm revisiting the decision to forcibly build deps and run `build_all` on a regular basis.  
I think it's unnecessary and clutters the runner work queue.  
We already build deps and Orca whenever we push or merge PRs into the main branch, so building it on a schedule doesn’t really add any value.

coryrc, 2025-12-23:
/assign-me