OrcaSlicer/OrcaSlicer #10417: "Allow only one OrcaSlicer instance" doesn't work on Linux

Matszwe02, 2025-08-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-alpha

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1 x86_64

### Printer

n/a

### How to reproduce

1. Download `OrcaSlicer_Linux_AppImage_Ubuntu2404_V2.3.1-alpha.AppImage`
2. open orcaslicer
3. select `allow only one orcaslicer instance`
4. try to open a file from terminal with command `orcaslicer xxx.stl`

### Actual results

```sh
$ ./OrcaSlicer_Linux_AppImage_Ubuntu2404_V2.3.1-alpha.AppImage 
[2025-08-15 13:22:35.273090] [0x00007096ad9fd4c0] [trace]   Initializing StaticPrintConfigs
dbus[2658181]: arguments to dbus_message_new_method_call() were incorrect, assertion "iface == NULL || _dbus_check_is_valid_interface (iface)" failed in file ../../../dbus/dbus-message.c line 1378.
This is normally a bug in some application using the D-Bus library.

  D-Bus not built with -rdynamic so unable to print a backtrace
Aborted (core dumped)
```

### Expected results

New OrcaSlicer terminal command should attach to an already existing OrcaSlicer instance, like it does well on windows. Linux users should also be able to use the `one orcaslicer instance` feature either with xdg-open or other ways.

### Project file & Debug log uploads

(I don't know if that log file is relevant)
```log
[error]	2025-08-15 13:29:13.627274[Thread 0x00007e0d6e7fc6c0]:Unable to open connection to spacenavd
[error]	2025-08-15 13:29:14.010472[Thread 0x00007e0e16f754c0]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2025-08-15 13:29:14.196080[Thread 0x00007e0e16f754c0]:01:29:14 PM: Navigation request to 'file:///tmp/.mount_OrcaSlHJDNhL/resources/web/homepage/index.html?lang=en_US' (target='')
[warning]	2025-08-15 13:29:14.199027[Thread 0x00007e0e16f754c0]:01:29:14 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2025-08-15 13:29:14.200321[Thread 0x00007e0e16f754c0]:01:29:14 PM: Navigation complete; url='file:///tmp/.mount_OrcaSlHJDNhL/resources/web/homepage/index.html?lang=en_US'
[warning]	2025-08-15 13:29:14.202615[Thread 0x00007e0e16f754c0]:01:29:14 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2025-08-15 13:29:14.203821[Thread 0x00007e0e16f754c0]:01:29:14 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2025-08-15 13:29:14.235366[Thread 0x00007e0e16f754c0]:01:29:14 PM: Script message received; value = {"sequence_id":1755257354,"command":"get_login_info"}, handler = 
[warning]	2025-08-15 13:29:14.235460[Thread 0x00007e0e16f754c0]:01:29:14 PM: Script message received; value = {"sequence_id":1755257354,"command":"get_recent_projects"}, handler = 
[warning]	2025-08-15 13:29:14.235534[Thread 0x00007e0e16f754c0]:01:29:14 PM: Script message received; value = {"sequence_id":1755257354,"command":"modelmall_model_advise_get"}, handler = 
[warning]	2025-08-15 13:29:14.236193[Thread 0x00007e0e16f754c0]:01:29:14 PM: Document loaded; url='file:///tmp/.mount_OrcaSlHJDNhL/resources/web/homepage/index.html?lang=en_US'
[warning]	2025-08-15 13:29:14.239662[Thread 0x00007e0e16f754c0]:01:29:14 PM: Warning: Error running JavaScript: Unsupported result type
[error]	2025-08-15 13:29:15.192271[Thread 0x00007e0e16f754c0]:calc_exclude_triangles:Unable to create exclude triangles
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jessyjones, 2025-08-16:
I don't know if the problem I'm having is related but when lauching from the file navigator using "open with" or double clicking if STL files are associated with OrcaSlicer, the program does open but is unable to import an additional object when opening via the same method if "Allow only one OrcaSlicer instance is checked". Drag and dropping the second file works as expected.

This is on Ubuntu, OrcaSlicer version 2.3.0

HoroTW, 2025-11-06:
@jessyjones That's most likely the same issue! The terminal just shows you "why" it doesn't work, but opening it in a file explorer essentially does the same thing.

Not many people seem to encounter this issue, I guess, because most don't activate the option. Even if they do, it still works with the first file they try to open, just not with the second.
I activated that option and ignored the behaviour for about a year because I did not know why it did not work, but since 'drag and drop' worked, it was not so important... 😅

@m404 had the same issue in https://github.com/SoftFever/OrcaSlicer/issues/2255