OrcaSlicer/OrcaSlicer #10421: Low Framerate on Orca Slicer 2.3.0 on Linux

citr0net, 2025-08-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Arch Linux 6.16.0-zen2-1-zen 

### Additional system information

Neofetch Output:
                  .o+`                   --------------- 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: MS-7D37 1.0 
               `+oooooo:                 Kernel: 6.16.0-zen2-1-zen 
               -+oooooo+:                Uptime: 5 mins 
             `/:-:++oooo+:               Packages: 1224 (pacman), 11 (flatpak) 
            `/++++/+++++++:              Shell: zsh 5.9 
           `/++++++++++++++:             Resolution: 1920x1080 
          `/+++ooooooooooooo/`           DE: Plasma 6.4.4 
         ./ooosssso++osssssso+`          WM: kwin 
        .oossssso-````/ossssss+`         Theme: Breeze-Dark [GTK2], Breeze [GTK3] 
       -osssssso.      :ssssssso.        Icons: Fluent-green-dark [GTK2/3] 
      :osssssss/        osssso+++.       Terminal: konsole 
     /ossssssss/        +ssssooo/-       CPU: 13th Gen Intel i7-13700F (24) @ 5.100GHz 
   `/ossssso+/:-        -:/+osssso+-     GPU: NVIDIA GeForce RTX 4060 
  `+sso+:-`                 `.-/+oso:    Memory: 6194MiB / 31873MiB 
 `++:.                           `-/+/

### Printer

All

### How to reproduce

Use the mesa drivers to launch orca slicer (env __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json orca-slicer)

When slicing, there will be a noticeably low framerate

### Actual results

Framerate can drop below 0 at times

### Expected results

No framerate drops

### Project file & Debug log uploads

Log File:
[warning]	2025-08-16 08:33:14.289047[Thread 0x00007fd0136f0cc0]:Preset already present, not loading: Creality K1C 0.4 nozzle
[error]	2025-08-16 08:33:14.910751[Thread 0x00007fd0136f0cc0]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2025-08-16 08:33:15.126665[Thread 0x00007fd0136f0cc0]:08:33:15 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2025-08-16 08:33:15.128910[Thread 0x00007fd0136f0cc0]:08:33:15 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2025-08-16 08:33:15.130121[Thread 0x00007fd0136f0cc0]:08:33:15 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2025-08-16 08:33:15.159599[Thread 0x00007fd0136f0cc0]:08:33:15 AM: Warning: Error running JavaScript: Unsupported result type
[error]	2025-08-16 08:33:16.263487[Thread 0x00007fd0136f0cc0]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2025-08-16 08:33:16.282202[Thread 0x00007fd0136f0cc0]:get_version, get_version not supported,return 00.00.00.00!
[warning]	2025-08-16 08:33:58.195208[Thread 0x00007fcf9d3fc6c0]:Arrange full params: {"min_obj_distance":0,"accuracy":1.000000,"parallel":1,"allow_rotations":0,"do_final_align":1,"allow_multi_materials_on_same_plate":1,"avoid_extrusion_cali_region":1,"is_seq_print":0,"bed_shrink_x":1.000000,"bed_shrink_y":1.000000,"brim_skirt_distance":0.000000,"clearance_height_to_rod":45.000000,"clearance_height_to_lid":101.000000,"clearance_radius":45.000000,"printable_height":250.000000,
[error]	2025-08-16 08:34:05.543616[Thread 0x00007fcf041a06c0]:call_process:got cancelled exception

Github prevented me from actually loading the project file. I uploaded it to here:
https://drive.google.com/file/d/1LHkFsB-W4XeLkJHoK09a5CQ-R1-kkNVe/view?usp=sharing

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

No other apps experience frame rate drops like Orca Slicer. This happens on all models, the dragon one was used to demonstrate this problem at its worst.

tool666schism, 2025-08-21:
Not sure if you ever figured this one out, but I've see the same issue on Fedora 42 with the flatpak verison of Orca slicer. Also on wayland.

Appimage version 2.3.0 runs perfect, but both flatpak version 2.3.0 and 2.3.1-alpha run terribly.

jondelh, 2025-08-24:
This does not happen when viewed through a web browser.My average framerate is 5 fps when using direct through OrcaSlicer but
about 15 fps in Firefox.

TomPcz, 2025-08-26:
> Appimage version 2.3.0 runs perfect,

Thank you! I downloaded the AppImage and it indeed works just fine (NVIDIA, Wayland, KDE, openSUSE Tumbleweed). As a bonus, dark theme works.

I wanted to try https://github.com/SoftFever/OrcaSlicer/pull/10313 which is supposed to fix some graphics issues, but that one does not start and crashed immediately.

closed 2025-08-27 by citr0net