OrcaSlicer/OrcaSlicer #10434: Segmentation Fault in App Build for Debian 13 and Arch AUR build

scotthep, 2025-08-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.31-alpha

### Operating System (OS)

Linux

### OS Version

Debian 13 and Arch Linux upto date rolling release

### Additional system information

CPU: AMD raven ridge
Graphics:   "
RAM 32GB

### Printer

Seg Faults on 1st run - never able to select printer

### How to reproduce

1.  In Debian:  Build 2.3.1-alpha "build_linux.sh -d" followed by "build_linux -s"
2.  In ArchLinux build this AUR [https://aur.archlinux.org/packages/orca-slicer-git](url)
3. Debian: Run ~/OrcaSlicer-Nightly-Builds//OrcaSlicer-nightly-builds/build/package/orca-slicer
4  In AUR: /usr/bin/orca-slicer

`add font of HarmonyOS_Sans_SC_Bold returns 1                                                                                                  
add font of HarmonyOS_Sans_SC_Regular returns 1                                                                                               
add font of NanumGothic-Regular returns 1                                                                                                     
add font of NanumGothic-Bold returns 1                                                                                                        
 
(orca-slicer:7961): Gtk-CRITICAL **: 13:01:53.603: gtk_cell_layout_get_cells: as                                                              sertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
.....
(orca-slicer:7961): Gtk-CRITICAL **: 13:01:54.467: gtk_window_resize: assertion                                                               'height > 0' failed                                                                                                                           
 
(orca-slicer:7961): Gtk-CRITICAL **: 13:01:54.473: gtk_widget_set_size_request:                                                               assertion 'width >= -1' failed
.....
(orca-slicer:7961): Gtk-CRITICAL **: 13:01:54.866: gtk_widget_set_size_request: assertion 'width >= -1' failed
 
(orca-slicer:7961): Gtk-CRITICAL **: 13:01:54.872: gtk_widget_set_size_request: assertion 'width >= -1' failed
 
(orca-slicer:7961): Gtk-CRITICAL **: 13:01:54.872: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault
~
`

### Actual results

Segmentation Fault

Am I missing a font package?

### Expected results

Should run

### Project file & Debug log uploads

Never get to the point of generating a project file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

scotthep, 2025-08-20:
There is either a redundant orca-slicer wrapper or the paths have changed.  In the working 2.3.0 in Debian 12, I start with the full path to ~/OrcaSlicer-2.3.0/build/package/bin/orca-slicer.  With export __EGL_VENDOR_LIBRARY_FILENAMES=/home/xxx/50_mesa.json followed by the full path to ~/OrcaSlicer-nightly-builds/build/package/bin/orca-slicer it still seg faults with the same sequence of warnings.
`xxx@XXXX:~/OrcaSlicer-nightly-builds$ ls
build                     deps            localazy.json    SECURITY.md
build_linux.sh            doc             localization     SoftFever_doc
Build.PL                  DockerBuild.sh  OrcaSlicer.mo    src
build_release.bat         Dockerfile      README.md        t
build_release_macos.sh    DockerRun.sh    resources        tests
build_release_vs2022.bat  flatpak         run_gettext.bat  tools
CLAUDE.md                 lib             run_gettext.sh   version.inc
cmake                     LICENSE.txt     sandboxes        xs
CMakeLists.txt            linux.d         scripts
xxx@XXXX:~/OrcaSlicer-nightly-builds$ ls build/
build_appimage.sh  cmake_install.cmake      CTestTestfile.cmake  package
build.ninja        compile_commands.json    orca_bu              resources
CMakeCache.txt     CPackConfig.cmake        orca-slicer          src
CMakeFiles         CPackSourceConfig.cmake  OrcaSlicer.tar
xxx@XXXX:~/OrcaSlicer-nightly-builds$ ls package/bin
ls: cannot access 'package/bin': No such file or directory
jsh@Cadd:~/OrcaSlicer-nightly-builds$ ls build/package
bin  orca-slicer  resources
xxx@XXXX:~/OrcaSlicer-nightly-builds$ ls build/package/bin
orca-slicer`

ionv3n0m, 2025-08-31:
I can confirm I'm also seeing this via AppImage and AUR install on Endeavour OS. 
 Endeavour OS (version 2025,03.19), linux firmware 6.16.3-arch1-1.

AppImage:
`'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Aug 30 20:11:53 timedout orcaslicer_main[347000]: invalid cast from 'wxPizza' to 'GtkCellLayout'
Aug 30 20:11:53 timedout orcaslicer_main[347000]: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Aug 30 20:11:53 timedout orcaslicer_main[347000]: invalid cast from 'wxPizza' to 'GtkCellLayout'
Aug 30 20:11:53 timedout orcaslicer_main[347000]: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Aug 30 20:11:53 timedout orcaslicer_main[347000]: Cannot register URI scheme wxfs more than once
Aug 30 20:11:53 timedout orcaslicer_main[347000]: Cannot register URI scheme memory more than once
Aug 30 20:11:53 timedout kernel: orcaslicer_main[347000]: segfault at 38 ip 00007f62cbc9f934 sp 00007ffd274be6a0 error 4 in libpangoft2-1.0.so.0.5600.4[b934,7f62cbc9c000+f000] likely on CPU 8 (core 10, socket 0)
Aug 30 20:11:53 timedout kernel: Code: 00 00 00 00 66 90 f3 0f 1e fa ff 25 16 15 01 00 66 0f 1f 44 00 00 55 48 89 e5 41 57 41 56 41 55 41 54 53 48 81 ec 88 00 00 00 <44> 8b 47 38 64 48 8b 04 25 28 00 00 00 48 89 45 c8 31 >
Aug 30 20:11:53 timedout systemd-coredump[347316]: Process 347000 (orcaslicer_main) of user 1000 terminated abnormally with signal 11/SEGV, processing...
Aug 30 20:11:53 timedout systemd[1]: Created slice Slice /system/systemd-coredump.
Aug 30 20:11:53 timedout systemd[1]: Started Process Core Dump (PID 347316/UID 0).`

AUR installation, via yay -S:

```
(orca-slicer-bin:762695): Gtk-CRITICAL **: 19:18:36.005: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:762695): Gtk-CRITICAL **: 19:18:36.007: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:762695): Gtk-CRITICAL **: 19:18:36.007: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:762695): Gtk-CRITICAL **: 19:18:36.008: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:762695): Gtk-CRITICAL **: 19:18:36.009: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer-bin:762695): Gtk-CRITICAL **: 19:18:36.010: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer-bin:762695): CRITICAL **: 19:18:36.239: Cannot register URI scheme wxfs more than once

** (orca-slicer-bin:762695): CRITICAL **: 19:18:36.239: Cannot register URI scheme memory more than once
Segmentation fault         (core dumped) orca-slicer
```

NateTheGrate, 2025-09-02:
Try doing the suggestions on the top comment. If those don't work I'm thinking its the latest NVIDIA driver that is being buggy so rolling back may also work.

NateTheGrate, 2025-09-02:
Ah even more interesting... that is good to know

scotthep, 2025-09-02:
@NateTheGrate In the initial post I also listed AMD graphics and in later posts indicated that I was running x11/OpenBox.  I got the same error message sequence -> segfault if I bypassed the wrapper script and ran the binary directly.  The redundant wrapper scripts are still there - perhaps the wrapper should be in the code directory as OrcaSlicer.desktop?

That said, the 01Sept2025 Nightly compiled in Debian 13 and spits out the same warnings  but does not segfault.  I have not tested it extensively most of my time has gone to importing my Sovol SV06Ace profile into Prusaslicer.  In contrast to this hot mess, typified by 3 weeks of pablum from the developers, PrusaSlicer code is clean, accepted by the Debian Project and available as a Debian Binary.

c2h5oh, 2025-11-01:
Aur package has received a bunch of fixes. Please confirm if current https://aur.archlinux.org/packages/orca-slicer works for you.

coryrc, 2025-12-25:
Build instructions for Linux: https://github.com/OrcaSlicer/OrcaSlicer/wiki/How-to-build#linux-instructions

Please build with `./build_linux.sh -est` (optionally adding -j or -r if needed). The `-e` gets you symbols.

Now run it like `gdb build-dbginfo/package/bin/orca-slicer`; type `start` and press enter once the prompt comes, then `cont` and press enter once the first breakpoint hits. Now do whatever causes the crash. Type `bt` and copy-paste the output into this Issue.

timnolte, 2026-02-06:
> Build instructions for Linux: https://github.com/OrcaSlicer/OrcaSlicer/wiki/How-to-build#linux-instructions
>
> Please build with `./build_linux.sh -est` (optionally adding -j or -r if needed). The [...]

@coryrc would be nice if that worked. :-/

```shell
./build_linux.sh -cdire
.....
./build_linux.sh: line 255: pushd: build-dbginfo: No such file or directory
```

coryrc, 2026-02-06:
> [@coryrc](https://github.com/coryrc) would be nice if that worked. :-/
> 
> ./build_linux.sh -cdire
> .....
> ./build_linux.sh: line 255: pushd: build-dbginfo: No such file or directory

`.....` you yada-yada'd the best part!

(aka you need to give me the full output if you want help)

coryrc, 2026-02-07:
I have an idea. Change line 255 from `pushd $BUILD_DIR > /dev/null` to:

```
  pushd "${SCRIPT_PATH}/${BUILD_DIR}" > /dev/null
```

It can't be worse than what's there.

coryrc, 2026-02-07:
Oh @timnolte you forgot the `-s`. That's what's wrong.

timnolte, 2026-02-07:
> Build instructions for Linux: https://github.com/OrcaSlicer/OrcaSlicer/wiki/How-to-build#linux-instructions
>
> Please build with `./build_linux.sh -est` (optionally adding -j or -r if needed). The [...]

@coryrc hmm, I'm building an AppImage, and that's intentional. I don't even think running it like you mention is going to work anyways.

coryrc, 2026-02-07:
> > Oh [@timnolte](https://github.com/timnolte) you forgot the `-s`. That's what's wrong.
> 
> [@coryrc](https://github.com/coryrc) hmm, I don't even see `-s` documented anywhere. What does that actually do? https://github.com/OrcaSlicer/OrcaSlicer/wiki/How-to-build#linux

Look again.

timnolte, 2026-02-07:
> > > Oh [@timnolte](https://github.com/timnolte) you forgot the `-s`. That's what's wrong.
> > 
> > 
> > [@coryrc](https://github.com/coryrc) hmm, I don't even see `-s` documented anywhere. What does that actually do? https://github.com/OrcaSlicer/OrcaSlicer/wiki/How-to-build#linux
> 
> Look again.

@coryrc I don't know what you are looking at, searching for `-s` on the entire page produces nothing. There are 2 examples that use it with other things but nothing about what it does on it's own.

[image]

coryrc, 2026-02-07:
You highlighted it there, "build OrcaSlicer with tests".

If you don't build OrcaSlicer, what will you put in an image?

coryrc, 2026-02-07:
> [@coryrc](https://github.com/coryrc) so just saying that I don't think that is what you think it is, and it also isn't documented on it's own because what is in the Docs say build it with tests (`-st`)

So why don't you run that?  (you'll need `-e` too)

> and the help output just says `-s: build the Orca Slicer (optional)` which even says it's optional.

It's optional if you only want to build the deps. But you don't.

> Which also the docs say for an AppImage you don't need `-s` all you need is `-i`. ;-)

Hopefully someone fixes it then. I've mostly given up fixing anything small because it never gets reviewed/approved i.e. https://github.com/OrcaSlicer/OrcaSlicer/pull/11885 which fixes one line in build_linux (err, again, because it got rolled back with other changes for a bad reason)

> <img alt="Image" width="897" height="127" src="https://private-user-images.githubusercontent.com/4312062/546469926-0ee37ebb-6247-408e-8994-88d60894a4d1.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NzA0MjQ3MDQsIm5iZiI6MTc3MDQyNDQwNCwicGF0aCI6Ii80MzEyMDYyLzU0NjQ2OTkyNi0wZWUzN2ViYi02MjQ3LTQwOGUtODk5NC04OGQ2MDg5NGE0ZDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MDIwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjAyMDdUMDAzMzI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDg4YzZjZjZiZjdjZDE3MTZjMDdiOTc1MzJkYjAwMWZiODMyNWE5MjkwMjgwMTc0ODI0OGM4NDFhZmNkNTM2ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.cJfjJB6FTzv_IKw_jnd9Ut52tcE_114gDPzKxyxYjws">

If you don't follow the steps listed, you won't get the outcome you want. If you find it confusing, please fix it. I didn't make most of it.

I gave a command to run, then you ran a different command, then you complained it didn't work. Why not run the command I gave instead of something else?

But, I actually missed a step in my command. The error would have been obvious to me, but oh well. Run this one:

`./build_linux.sh -deirst`

I even added an `-i` for you