OrcaSlicer/OrcaSlicer #10465: Exclusion area Kobra 3 alpha 2.3.1

lordflyingpoo, 2025-08-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Alpha 2.3.1

### Operating System (OS)

Linux, Windows

### OS Version

Windows 11, Ubuntu 24.04

### Printer

Anycubic Kobra 3

### How to reproduce

1. Select Anycubic Kobra 3 as printer.
2. Add model to plate.

### Actual results

Error:
tinker.stl is too close to exclusion area, there may be collisions when printing.

[image]

### Expected results

Be able to slice plate but it will not.

### Project file & Debug log uploads

[tinker.zip](https://github.com/user-attachments/files/21881197/tinker.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

MJFairbairn, 2025-08-20:
Have same problem on my K3 and with the Beta versions as well.

Phenix54s, 2025-08-21:
same thing , exclusion area looks inside out 

2.3.1 alpha on windows11

wspock, 2025-08-22:
Same here 

[image]

jmmerida, 2025-10-05:
Thanks for the fix. But I still have this issue:

[image]

I can just ignore it. But seems odd.

dolence, 2025-10-07:
> I can just ignore it. But seems odd.
Same here!

zekrom-vale, 2026-03-06:
I have the issue with Kobra 3 on Ubuntu 24 with version 2.3.1.  But no setting other than ` ` (none) caused it to think everything is an exclusion zone, kicking things off the plate or giving gcode errors/warnings.

kameqdev, 2026-10-02:
even in case the excluded zone is displayed correctly (like when using `0x0, 255x0, 255x255, 0x255, 0x0, 2x2, 2x253, 253x253, 253x2, 2x2`) and allows slicer to slice a model without errors, there is an issue with auto arrange objects option, that puts all models outside the plate