OrcaSlicer/OrcaSlicer #10468: adding a second paint trace to enclose scarf seam

valadvd-art, 2025-08-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

as also scarf seam leave slight signs at starting and ending point it would be great to have an option to draw also the stop line to enclose scarf seam, for example beetween two edges to make the seam even less noticeable.

This is the actual behavior:

[image]
[image]

and this is the feature to rerpoduce:

[image]
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

add a choice to paint the end of scarf seams:

[image]
[image]

valadvd-art, 2025-08-21:
thanks DIscip for reply. I thought the conditional scarf was about the slicer choosing where to put the seam in an automatic disposition (where to use scarf and where to use normal) but you're right, i'll give it a try. Thanks again 😊

valadvd-art, 2025-08-21:
I tried now and the conditional option change where to apply scarf and where to apply normal, not a selective ending of the scarf joint 😞

discip, 2025-08-21:
**`Scarf joint seam`** is best for hiding seams on smooth surfaces, mostly on parts with round features.

On surfaces with edges, like the one in your picture, the seam can easily be hidden in the inner corner without the need for a **`Scarf joint seam`** — hence the recommendation. 😊

However, if you want the walls to be watertight, you could activate **`Staggered inner seams`**.

valadvd-art, 2025-08-21:
Thanks Discip for your time. i agree with you that seam can be hide in corner, i don't have a picture of piece in the project i send, i tried both on inner and on outer corners but, when the corners are near to 90 degree or sharper the seam is almost invisible as in pictures below

[image]
[image]

but when angle start to enlarge the seam is very visible:

[image]

instead with scarf i succed in better hiding seam (i've to work on stringing😅)

[image]

the problem is that, since the two edges are not parallel to each others, scarf ends after and leave a sign 

[image]

[image]

I've tried using modifiers to delimit scarf but they split object and the result is worse.
The aim is to make a scarf closed between two edges regardless of their mutual inclination

discip, 2025-08-21:
Ah, thanks for the insight. 👍
Now I understand.
However, I think it wouldn't be easy to implement.

valadvd-art, 2025-08-21:
thanks again for your time!... i hope that this feature, even if not easy to implement, can be taken into consideration🙂