OrcaSlicer/OrcaSlicer #10470: Wipe tower ignores selected filament in 'Filament for Features' (regression: 'Soluble material' workaround no longer works in 2.3.1 alpha)

LOKAT24, 2025-08-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 alpha (regression from 2.3.0)

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

None – they don't matter.

### Printer

Prusa XL 2T

### How to reproduce

1. Open a multi-material project in OrcaSlicer (tested on Prusa XL 2T).
2. In Filament for Features, set 'Wipe tower' to a single material (e.g. PETG).
3. Assign soluble/support material to the other tool (see screenshot).
4. Slice the project.
5. Observe the wipe tower in Preview mode: check the shell/perimeter color/material.

### Actual results

Despite selecting a single material for 'Wipe tower', the wipe tower's shell/perimeter is still printed using all active materials, as shown in Preview. In OrcaSlicer 2.3.0, I could work around the broken 'Wipe tower' filament mapping by assigning the second material (e.g. PLA) as 'soluble material'. This prevented it from being used for the wipe tower shell. In 2.3.1 alpha, even this workaround no longer works: the 'soluble material' is now used for the wipe tower shell regardless, and there is no way to force the tower to use just one filament.

See the attached images for comparison between 2.3.0 and 2.3.1 alpha:

[image]
[image]
[image]

### Expected results

Please fix the logic for assigning the wipe tower shell material so that the selected filament in 'Filament for Features → Wipe tower' is always respected and only that material is used for the wipe tower shell (and optionally surfaces). If this cannot be fixed soon, at least restore the previous behavior from 2.3.0, where marking a material as 'soluble' excluded it from wipe tower shells. Currently, there is no way to control the wipe tower shell material in multi-material prints.

### Project file & Debug log uploads

This is 3MF.

[WipeTowerProblem.zip](https://github.com/user-attachments/files/21888566/WipeTowerProblem.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

This appears to be a regression from OrcaSlicer 2.3.0 (see screenshots for difference). Previously, the inability to control wipe tower shell material could be bypassed by setting the secondary material to 'soluble', which excluded it from the wipe tower. In 2.3.1 alpha, this workaround no longer works—'soluble' materials are now used for the wipe tower shell as well. This change removes all user control over wipe tower shell material in multi-material prints.

Printer: Prusa XL 2T, AMS/MMU disabled, multi-material setup with PLA and PETG. See screenshots for settings and wipe tower behavior. Let me know if more info is needed or if you want the 3MF project and logs attached here.

nic335, 2025-08-24:
I can also confirm that issue!

traxman25, 2025-08-24:
Shouldn't even have to set the filament type to work around the main bug. 

In reality the main bug here is that the wipe tower wall extruder selection (and all of the other selections in this area; Wall, infill, and solid infill) doesn't work no matter what. This really needs to be fixed because this makes Orca useless for multi material printing.

SoftFever, 2025-09-09:
I'll take a look

jwolv, 2025-10-03:
> Shouldn't even have to set the filament type to work around the main bug.
>
> In reality the main bug here is that the wipe tower wall extruder selection (and all of the other selections in this area; [...]

Can confirm in OrcaSlicer 2.3.0, all of the "Filament for Features" options were ignored regardless of the base material of the object, or any painting done - though I never did try the "soluble materials" selection for any filament profile.

I updated to OrcaSlicer 2.3.2 today to see if this was fixed, but it has the same issue - came looking for similar post before creating a new issue, so commenting to add my +1 and share my testing

0x0nium, 2025-11-02:
I hope they fix it soon — it's the only thing keeping me from switching away from Orca and PrusaSlicer.
I can confirm the same issue on version 2.3.2.