OrcaSlicer/OrcaSlicer #10474: New calibration models are'nt suitable for spiral mode

CodeZ1LLa, 2025-08-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

V2.3.1-alpha

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3 Max, 0.4 nozzle

### How to reproduce

1. Select Calibration
2. Select cornering or input shaper (any calibration that uses spiral mode)

### Actual results

When sliced, it leaves the holes even with the tiniest wall width that 0.4 nozzle can print

[image]

### Expected results

solid walls

### Project file & Debug log uploads

[OrcaSlicer.zip](https://github.com/user-attachments/files/21899668/OrcaSlicer.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

ianalexis, 2025-08-20:
> I just realized that the issue is a smooth spiral mode that is ON by default

That's on purpose. I'm sure we've seen this before and solved it somehow. Let me check it out.

ianalexis, 2025-09-14:
> I can not upload the project file, so I renamed it to .doc.
> 
> Rename it back, please
> 
> [Input shaping Frequency test.doc](https://github.com/user-attachments/files/22307615/Input.shaping.Frequency.test.doc)

Excelent, will take a look. At first i think there is something wrong with your machine setting.

closed 2025-09-28 by SoftFever