OrcaSlicer/OrcaSlicer #10476: SIGSEGV shortly after startup

BekaValentine, 2025-08-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Linux

### OS Version

Arch

### Printer

Bambu X1C

### How to reproduce

Start Orca Slicer.

### Actual results

UI loads but most of the time is non-interactive. SIGSEGV happens shortly thereafter.

```
fish: Job 1, 'orca-slicer-bin' terminated by signal SIGSEGV (Address boundary error)
```

### Expected results

Orca Slicer is usable.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Previously worked, but then suddenly started getting hit with SIGSEGV.

peterzuger, 2025-08-20:
Getting the same issue with 2.3.0 on Arch linux.

The segfault occurs in libpangoft2-1.0.so.0 downgrading to pango-1:1.56.3-1 fixes the issue.

BekaValentine, 2025-08-20:
Following @peterzuger's recommendation, I downgraded `pango`, and that fixed it. The first thing that showed up upon starting, which differs from the `SIGSEGV` state, is the version update window. After denying an upgrade to 2.3.0.1-alpha, it ran fine. I then exited, moved `pango` back to the most recent version, and now Orca Slicer has no problems.

It seems to be the case that this version update window is triggering some problems.

n0xt3ch, 2026-02-07:
Cachyos + i3wm - orca-2.3.2-dev

After a couple of time, I decided to add xfce (main is i3wm) as WM and orca starts without issues. So I removed xfce again and found via trial and error, that installing **gst-plugins-base** and **gst-plugins-good** helped out. Orca starts without any problems.

duub, 2026-04-30:
is this something it should be fixed or it's impossible to make it work with all distros?

LukaCek, 2026-09-04:
Another data point for this same underlying Pango/FreeType crash, with a different trigger and a symbolized backtrace that pinpoints the mechanism.

**Environment:** Arch (Omarchy/Hyprland), OrcaSlicer 2.4.2-2 (AUR), freetype2 2.14.3-1, cairo 1.18.4-1, pango 1.58.2-1, gtk3 3.24.52-1.

**Trigger:** SIGSEGV while GTK is presenting a **native file dialog** (`gtk_native_dialog_run` → `gtk_widget_show` → realize), not at startup or in the update checker. It crashed during widget size measurement for the dialog.

**Backtrace (symbolized, frame 0 is inside orca-slicer):**
```
#0  orca-slicer        (crash, null deref)
#1  FT_Set_Var_Design_Coordinates      libfreetype.so.6
#2..#6  cairo ... cairo_scaled_font_glyph_extents   libcairo.so.2
#7..#15  pango ... pango_layout_get_size             libpango
#16..  gtk_widget_get_preferred_width (recursive size-measure chain)  libgtk-3
#105  gtk_native_dialog_run → gtk_widget_show → realize
```
Kernel: `segfault at 0 ... error 4` (read from NULL). `si_code=SEGV_MAPERR`.

**Root cause / fix (confirmed):** Pango tries to resolve the Korean label "한국어" (language selector) to the **NanumGothic** font; when `NanumGothic` isn't installed, Pango returns a null font handle to FreeType's variable-font code (`FT_Set_Var_Design_Coordinates`), which dereferences it. Installing `ttf-nanum` from the AUR (provides `NanumGothic.ttf`) stops the crash entirely. `fc-list | grep -i nanum` was empty before install, present after.

So: same class of bug as #10476/#10524, triggered on the file-dialog path, and the general `ttf-nanum` workaround holds. Worth fixing inside Orca by guarding against the null font handle rather than relying on a font that isn't always installed.

> Filed by big-pickle via opencode.