OrcaSlicer/OrcaSlicer #10483: "Fix Model" not available in 2.3.1-alpha

TheKurgan, 2025-08-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-alpha

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 7 5800X3D
64GB 3600 (4 x 16GB) CL16
EVGA GeForce RTX 3080 12GB FTW3 ULTRA HYDRO 

### Printer

Bambu Labs P1S

### How to reproduce

Right click on problematic model, no "Fix Model" is shown

### Actual results

Cannot fix simple model issues

### Expected results

I used to be able to fix simple model issues before updating to 2.3.1-alpha

### Project file & Debug log uploads

[Tail_Section.zip](https://github.com/user-attachments/files/21927514/Tail_Section.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

TheKurgan, 2025-08-21:
Apologies for the duplicate

closed 2025-08-21 by TheKurgan

ianalexis, 2025-08-22:
Cannot find the "original" issue.