OrcaSlicer/OrcaSlicer #10485: "Replace With STL" does nothing

Sod-Almighty, 2025-08-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-alpha

### Operating System (OS)

macOS

### OS Version

MacOS 12.6.4

### Printer

Bambu Lab P1S

### How to reproduce

1. Update to 2.3.1-alpha
2. Right-click an object.
3. Select "Replace with STL".
4. Choose a STEP file (which always worked before).

### Actual results

Nothing happens. The object is not replaced by the STEP file.

### Expected results

The object should be replaced by the selected STEP file.

### Project file & Debug log uploads

[large drawer v39.3mf.zip](https://github.com/user-attachments/files/21928635/large.drawer.v39.3mf.zip)

[logs.zip](https://github.com/user-attachments/files/21928644/logs.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Zogar89, 2025-08-22:
I tried it with alpha and in my project this works well.

tlhintoq, 2025-08-30:
Are you updated the STEP file then trying to update the Orca project?
If that's the case i think you want to use "Reload from disk".
Replace (I think) is expecting a new model on a new pathname.  If you updated the file then 'replace' doesn't see it as a new item.

trx1, 2025-10-08:
Same here. Renaming the new STEP file to correspond with the original and attempting to reload it from disk does not appear to be working either. I use the replace quite often when I'm making design tweaks and printing test parts.

Windows 11
Version 2.3.1 release (build f2d10a0)

Sod-Almighty, 2025-12-11:
Four months now and it's still broken...

closed 2025-12-14 by SoftFever