OrcaSlicer/OrcaSlicer #10508: [Feature Request] Ironing settings should be part of the filament profile

glaucon1984, 2025-08-24:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Ironing settings should be configured at the filament profile level, it doesn't make sense to save a project with specific settings for a regular PLA when another user might decide to use Matte PLA or PETG instead. The ironing should be a setting independent of the model, the same way the bed temperature is, and depend on the type of filament used.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

- Allow to store ironing settings as part of the filament profile.
- Ironing will automatically apply the ironing settings stored in the profile depending on the filament selected.
- (Maybe harder to implement) This could enable multi material parts (for example, basic PLA + matte PLA) to have correct ironing on all sections of the print

### Additional context

Ironing is a great feature that gives a massive quality to the prints, but it's rarely used because it's very hard to keep track of the best settings for each material. A 3MF with ironing settings baked in will only produce a good result if the material matches perfectly. Users and filament manufacturers can tune their filament profiles for the best result, or even alert when enabling this option if the material has known issues with ironing.

ianalexis, 2025-08-30:
Will end some PRs im working on and if nobody else take this i will.

ianalexis, 2025-09-01:
/assign-me