OrcaSlicer/OrcaSlicer #10534: Can't send prints to X1c in LAN mode, Sending a print prompts to connect to printer despite printer being connected

kekivelez, 2025-08-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

Windows 10/11

### Additional system information

CPU: 13th gen intel i9 13900k
Memory: 32GB 
Graphics: Nvidia 3080

### Printer

Bambu Lab X1c

### How to reproduce

1. Slice print
2. Click Print
3. Prompted to connect to printer
4. Input IP and access code hit connect
5. Prompted to give printer a name and input SN
6. Input Name, SN, printer model, hit connect

### Actual results

After step 6 the window for printer name/SN (step 5) closes and returns me to the sliced print view and the loop starts back at 1 (ie. hit print, be asked to connect, etc)

### Expected results

Print is sent to printer and print starts

### Project file & Debug log uploads

I tried a fresh install of orca so i only had 2 logs in my log dir, a cleartext and a log.enc. No particular project attached as this was happening regardless of print
Clear text:
```
[warning]    2025-08-19 01:41:12.682241[Thread 0x00000a7c]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]    2025-08-19 01:41:12.682241[Thread 0x00000a7c]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf
[warning]    2025-08-19 01:41:16.006938[Thread 0x00000a7c]:1:41:16 AM: Navigation request to 'file:///C:/Program%20Files/OrcaSlicer/resources/web/homepage/index.html?lang=en_US' (target='')
[warning]    2025-08-19 01:41:16.016938[Thread 0x00000a7c]:1:41:16 AM: Document loaded; url='file:///C:/Program%20Files/OrcaSlicer/resources/web/homepage/index.html?lang=en_US'
```

log.enc: [debug_network_Tue_Aug_19_01_41_12.log.zip](https://github.com/user-attachments/files/21997634/debug_network_Tue_Aug_19_01_41_12.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

I am using Orca slicer 2.3 (build 70931e5), when i try to send a print to my LAN mode printer it prompts me for the access code, after entering it it prompts me for my device name model and serial number, after I input that the window closes without error but sends me back to the print confirmation screen... rinse and repeat. It doesn't let me move past those windows (See screenshots). 

My printer is on FW version 1.07.04, and I can see on the printer tab that the device is connected, I can see the AMS filament data as well as control the on device camera

[image]
[image]
[image]
[image]

skeet70, 2025-08-28:
I'm experiencing this same problem with a P1S. Firmware is 01.08.02.00. I can also see the AMS, control printer settings, view camera.

Initially hitting send gave me SSL errors a few times, then started giving me the connect pop up every time I hit send even though it's connected.

tlhintoq, 2025-08-30:
I was getting that a lot too lately.
What worked for me... 

LogOut from Bambu on the [HOME] screen.
Exit Orca

Download and install the latest BambuStudio (and thus its network component)
Run it, log in, connect,  send a tiny job just to prove it.
Exit BambuStudio

Now run OrcaSlicer. LogIn. Produce and print a job etc.