OrcaSlicer/OrcaSlicer #10551: Allow Filament to be specified on a per-plate basis

RayInNoIL, 2025-08-27:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Printing a project that uses different filaments, that need different speeds/temperatures (e.g. Stone PLA and Silk PLA), takes extra steps because the filament setting needs to be changed, and the project/plate(s) need to be re-sliced before printing.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Allowing filament to be set on a per-plate basis would be helpful.  This way, all plates in a project could be kept together, all the items printed in each filament could be printed on one plate, and you would not have to edit the filament setting and re-slice if each plate needs a different temperature, max speed, etc.

### Describe alternatives you've considered

1. Set the filament for the first plate.
2. Print the first plate.
3. Set the filament for the second plate.
4. Print the second plate.
5. Repeat Steps 1 & 2, for each additional filament
6. Start all over again at Step 1, the next time you chose to print the project

pm0u, 2025-08-27:
This would be a great feature. Printing Voron parts or Hula feet I found myself wanting to group as a project but there were often different filament requirements for different parts of the project, leading to a long list of filaments that led to (ultimately ignored) filament change commands.

LupusTheCanine, 2025-09-17:
I think it would be cool if we could have subprojects with independent settings. It is quite common for large projects to require groups of components to be printed from different materials and different settings.

j3kestrel, 2026-09-01:
I am expanding the scope of this issue, but it may be worth while when considering what an implementation might look like.  I would very much like to be able to select a filament and process settings (as a saved group, not setting each item one at a time under "objects")  at both the plate and object level.  I am not using a multi-filament printer, but may want to have, for example, different cooling settings for the same single filament on the same plate.  In general, it feels like there is good support for multi-color cosmetic objects, but not so much support for multi-material and diverse processes selection for functional printing.

j3kestrel, 2026-09-01:
Found a related but ignored issue: https://github.com/OrcaSlicer/OrcaSlicer/issues/7238

A couple of related requests on reddit:

https://www.reddit.com/r/OrcaSlicer/comments/1dyjn71/is_it_possible_to_have_different_print_settings/
https://www.reddit.com/r/OrcaSlicer/comments/1ifamuu/changing_filament_settings_per_object/