OrcaSlicer/OrcaSlicer #10575: 2.3.1 alpha weird toolpath

flint723, 2025-08-30:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1

### Operating System (OS)

Windows

### OS Version

windows 11

### Additional system information

14th gen intel 14600k
48gb ddr5
3080ti gpu

### Printer

Qidi Plus4

### How to reproduce

Print anything with larger/long and thin supports.

### Actual results

The support first layer will be generated in a weird pattern, not always drawing border first and even if it does, the infill may not be sequential and jump around, starting and stopping.

### Expected results

It should draw support borders first and just fill in the infill from end to end without breaks.  The file is a sliced gcode (needs .zip changed to .gcode) which isn't that bad but still jumps around and doesn't draw border first.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

I tried to send files but it said 3mf not supported, .gcode not supported.  So this .zip needs to be renamed to .gcode to see it.  

[Qidi Plus4 Riser Back Left_PETG-CF_2h32m.zip](https://github.com/user-attachments/files/22056535/Qidi.Plus4.Riser.Back.Left_PETG-CF_2h32m.zip)

flint723, 2025-08-30:
To add a little bit, in qidi studio this generates a border and fills in the infill end to end without breaks.

flint723, 2025-09-03:
@SoftFever Here are those things.  Model is public, no big deal. 

[Qidi Plus4 Riser Back Left 3mf and stl.zip](https://github.com/user-attachments/files/22127893/Qidi.Plus4.Riser.Back.Left.3mf.and.stl.zip)

inigomontoyaik, 2025-09-04:
ive got a similar problem, but its only in some of the support contact points.

[image]

[image]

[multifan3 v10 combo gcode.txt](https://github.com/user-attachments/files/22140605/multifan3.v10.combo_PETG_0.4mm_1h41m.code.txt)

SoftFever, 2025-09-06:
Hmmm, that's strange.
I downloaded the build and it works fine on my machine now.
Are you using Windows or Mac?
https://github.com/user-attachments/assets/0c379352-960f-4659-a5da-e2dbfc93220e

flint723, 2025-09-06:
Windows 11.  I went back to 2.3.0 and it works as it should.

Just now went back to this build.  I see you used Tree supports, for me tree works as it should (as yours does).  Normal support does not, it still jumps around on the first layer infill.  Though as I said the border is at least drawn first with this build.

flint723, 2025-09-06:
Even very small supports do it.

https://github.com/user-attachments/assets/ead34845-6b27-4c49-bf86-137bd175603d

flint723, 2025-09-06:
Well, I was wrong about tree supports working correctly, they also will sometimes not fill sequentially.  Just depends on the model.

SoftFever, 2025-09-07:
Yes, I figured it's regression caused by this PR(https://github.com/SoftFever/OrcaSlicer/pull/9407)

closed 2025-09-07 by flint723

flint723, 2025-09-16:
The main problem (support first layer infill not going down sequentially) is fixed but you are correct, in 2.3.1 beta the support first layer still isn't quite the same as 2.3.0.  It sometimes will put the infill down first and draw the outer wall after.  Unless it's doing something different now for you.

I made a different issue, #10711