OrcaSlicer/OrcaSlicer #10579: 3D Exclusion Zones

pm0u, 2025-08-31:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When using an toolchanger on a 2.4 or theoretically any printer with a flying gantry, you lose some Y volume however only at certain Z heights.

Currently, to work around this lost people either write off the front part of the plate (leading to significant build volume loss that is technically usable) or do their best to plan a print so that it can utilize the available volume via manual assessment (which is obviously prone to human error).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be great to be able to assign both 2D (assumed to apply at all build heights) and 3D (defined in x,y,z coordinate tuples) exclusion zones so that the slicer could handle this and ensure that things like travel don't enter the exclusion zone. This may require a new syntax as determining a "closed" polygon in 3 dimensional space will be a bit more complicated.

### Describe alternatives you've considered

1. Exclude the zone at all heights, this may reduce a Voron 350 to more like 350x290x350
2. Plan well and examine prints and travel paths carefully to maximize build volume. Many people use a custom build plate image to remind them of maximum heights in certain areas.

### Additional context

This is from the stealthchanger FAQ and visually describes the problem.

[image]

https://github.com/DraftShift/StealthChanger/wiki/Support-and-FAQs#how-to-calculate-build-loss

pm0u, 2025-09-01:
> > Exclude the zone at all heights, this may reduce a Voron 350 to more like 350x290x350
> 
> This alternative is already available in the printer preset. For example, this is area excluded for Bambu X1/P1 printers by default because head movement into this region operates the filament cutter.
> 
> <img alt="Image" width="965" height="367" src="https://private-user-images.githubusercontent.com/139707784/484028124-415249fe-2822-43a7-8afa-052d982e2a4e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTY2OTQ0OTAsIm5iZiI6MTc1NjY5NDE5MCwicGF0aCI6Ii8xMzk3MDc3ODQvNDg0MDI4MTI0LTQxNTI0OWZlLTI4MjItNDNhNy04YWZhLTA1MmQ5ODJlMmE0ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkwMVQwMjM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNzg2MWU2MDBkZTQ3MjExMmM5ZDM0NWY0ZGRkYzIyMGI1Yzc0Yzc2MjAzYjg1ZjE3ZTVkOWMwYWI1NTdkZTEzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.v8LYrjkXZQB73zLRTo0fH3cBKvnv3_p4Et3gNWdmbu4">

I'm aware this exists, that's why it's listed under alternatives I've considered. I am hoping for a better solution since in this case following that path creates a massive exclusion zone. thank you though.

MHillyer, 2025-09-25:
Something I would also support seeing, flying gantry toolchangers like Stealthchanger only lose space at specific Y heights, so using the current exclusion area implementation takes away space that could be preserved, especially since a lot of prints will not be the combination of height/width to enter the collision area, but many will be wide enough without being high enough, making it impractical to use the current 2D exclusion area.

closed 2026-10-22 by SoftFever