OrcaSlicer/OrcaSlicer #10581: Fan speed does not respect layer time...not interpolating properly.

PhilBaz, 2025-08-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

10

### Additional system information

AMD poweful

### Printer

Voron 2.4 and others

### How to reproduce

Set layer time to 7 second (the actual layer time for the cube) fan speed to 10%.

### Actual results

Fan speed is over 50%, but should be 10% since the layer time for the object and 'min fan speed / layer time' threshold are the same.

[image]

[image]

### Expected results

Fan speed should be 10% since layer time for the object and 'min fan speed / layer time' threshold are the same.

### Project file & Debug log uploads

[fan speed layer time disconeted.zip](https://github.com/user-attachments/files/22061982/fan.speed.layer.time.disconeted.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Thanks...

MxBrnr, 2025-08-31:
Duplicate of #7980

pm0u, 2025-08-31:
Duplicate, but still a problem. we should either reopen the other or keep this one?

PhilBaz, 2025-09-01:
@MxBrnr Can someone @ the actual devs.... This is primary functionality breaking. If its not rude we should draw their attention to this. 

Is this buggy in Bambu studio as well?

PhilBaz, 2026-10-31:
Has anyone had a chance to look at this....Its a bug yeah?

closed 2026-11-15 by ianalexis