OrcaSlicer/OrcaSlicer #10582: Cant add more than one  height range modifier also cant delete it if you create,

raidycv, 2025-08-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1

### Operating System (OS)

macOS

### OS Version

lates mac os 15.6

### Printer

voron trident/ bambulab a1 

### How to reproduce

Can't add more than one  height range modifier, also can't delete it after creating 
 The only workaround i found to create more than one is to create different layer height range modifiers in different objects, copy them, and paste in the same objects, so it can work, but it is so unnecessary.  

### Actual results

can only add one height range modifier

### Expected results

when click in height range modifier should be suposed to add more than one time 

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/user-attachments/files/22062151/Cube.3mf.zip)

[video repro 1.mov.zip](https://github.com/user-attachments/files/22062153/video.repro.1.mov.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2026-02-23:
Nice to know where those buttons are - Thanks for that.  I have Orca spanning two monitors vertically. Those buttons way way way down at the bottom of the second monitor just wouldn't have been seen.

That said: Its still a bug that using the context menu on the option and selecting "add height range modifier" really should do the add.  Not working really does feel like this is still a bug and probably fairly low-hanging fruit for someone looking to make a fix.

Finnitio, 2026-03-18:
That was so hard to find :D. Should also work with the context menu as it is the same functionality

chi1l1l, 2026-09-04:
The -/+ buttons look like they would adjust the range in some way with how they're currently laid out. Unless the + buttons insert modifiers in a specific order, they could be replaced with a blank row underneath. That way, each Height Range row just has a - button, and the blank bottom row has a +, making the feature intuitive.

When right clicking the object, the Height Range Modifier doesn't match any other options in the section. You're led to assume the button means 'Add Height Range Modifier', since it shares a section with 'Add Part', 'Add Blocker', etc. Currently, it just takes you to the Layers view, which is easier to do without the button. Renaming this to 'Add Height Range Modifier' and having it go to Layers while adding a new modifier would make it feel more natural.

Kind of repeating what's been said, I'm just leaving my thoughts here since I had this issue as well. The buttons are right there but went entirely misunderstood, so it does feel more like a UI/UX 'bug' rather than a new feature to request.