OrcaSlicer/OrcaSlicer #10584: Ironing Speed Selection not available

flyingcelt, 2025-08-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ellegoo Centari Carbon

### How to reproduce

No option to select Ironing speed once Ironing is selected. Only Ironing Flow is available. This is not the case when using version 2.3.0. With current printer.

### Actual results

Only work around was to revert back to version 2.3.0

### Expected results

Using 2.3.0 able to select Ironing Speed

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kristal440, 2025-10-18:
I don't have it on either of the locations

[image]

[image]

> <img alt="Image" width="400" height="455" src="https://private-user-images.githubusercontent.com/139707784/490850673-b3756739-4a23-4fdb-9e25-7707765fb35d.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjA3ODQzOTEsIm5iZiI6MTc2MDc4NDA5MSwicGF0aCI6Ii8xMzk3MDc3ODQvNDkwODUwNjczLWIzNzU2NzM5LTRhMjMtNGZkYi05ZTI1LTc3MDc3NjVmYjM1ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMDE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTAxOFQxMDQxMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MjdkZjdmNjUwZDhhMzBiNDdmZTcyZjI2N2Q5YjJlODA0MzZiMDNmNDk2OGEyODUxNzBiOTA0M2FhODA2N2RhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.S8ubiu_vLc9wM2hCwOYFp3-mQqmMediu2GB9bg0btTc">

LJKraus, 2025-10-18:
Works fine here, Windows release 2.3.1, build f2d10a0.

Ironing must be enabled on the Quality tab **before** Ironing speed will be available in the Speed tab.