OrcaSlicer/OrcaSlicer #10585: Readable flushing volumes table

0verEngineer, 2025-08-31:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Can we get a usable/readable flushing volumes config table like Bambu Studio?

Bambu Studio:

[image]

Orca Slicer:

[image]

In the Orca Slicer version it is not possible to read values from 100 +, all the "35" entries you see in the table are actually "350"

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

.

jldavis1969, 2025-11-30:
> I am using Linux 6.17.5-arch1-1, gnome and Orcaslicer 2.3.0 build [70931e5](https://github.com/OrcaSlicer/OrcaSlicer/commit/70931e5321fa66966a5bfb251efca0e82307d427), and I have this also. What version did this fix ended up in (I don't know how to find out that myself), or are planned for?
> 
> <img alt="Image" width="568" height="195" src="https://private-user-images.githubusercontent.com/13905461/514744380-f4b4640e-9600-48d7-a919-6bc48ea775f0.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ0ODU1OTMsIm5iZiI6MTc2NDQ4NTI5MywicGF0aCI6Ii8xMzkwNTQ2MS81MTQ3NDQzODAtZjRiNDY0MGUtOTYwMC00OGQ3LWE5MTktNmJjNDhlYTc3NWYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTMwVDA2NDgxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY5MDNhZmEyNjJkZjlhZjJlOTE1OWIyZjc1MGNhOGE4OGUxMjUzZGM2M2JhOGE5NmFjOThjYzY1M2ZmYmEzNzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.OqoFckLZ_3QrirPGeW0AuyDPqzBwmlk0jvU9WBhIi6g"> <img alt="Image" width="394" height="496" src="https://private-user-images.githubusercontent.com/13905461/514744399-d0c9182e-5332-4021-b464-c3786d138312.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ0ODU1OTMsIm5iZiI6MTc2NDQ4NTI5MywicGF0aCI6Ii8xMzkwNTQ2MS81MTQ3NDQzOTktZDBjOTE4MmUtNTMzMi00MDIxLWI0NjQtYzM3ODZkMTM4MzEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTMwVDA2NDgxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNkMzE1ZjBmNGZiNzM4YjVjYTk3MjA0NjUxMTM2MDVjMDRlYWY5ZmVjNWEyNTY3MTBjMzA0ZGUwNzFhMjIzZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.RRisxyEeE3zIAhG2ELUmk3V3nc5X3A3T9_P1bi2aHVA">

Question: How does the Flushing Volumes Multiplier affect the numbers? Say you have a value of 100, and a multiplier of 0.50. Does it mean the actual number is 50? In the picture above you have from Black (line 1) to White (line 2) with a value of 50 and the multiplier is 0.80. Is that 80% of 50?

jldavis1969, 2025-11-30:
Thank you for your timely reply. Now I know how to set the numbers.