OrcaSlicer/OrcaSlicer #10591: Error when attempting to upload files to 3D printer

dgmoney43, 2025-08-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Neptune 4 Max

### How to reproduce

Upload and print

### Actual results

Error uploading to print host:
HTTP 502: <html>
<head><title>502 Bad Gateway</title></head>
<body bgcolor="white">
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx/1.14.2</center>
</body>
</html>

### Expected results

Begin Print

### Project file & Debug log uploads

[debug_Sun_Aug_31_19_32_47_28832.log.zip](https://github.com/user-attachments/files/22069259/debug_Sun_Aug_31_19_32_47_28832.log.zip)

[swolepikachu_full.zip](https://github.com/user-attachments/files/22069269/swolepikachu_full.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

For context, this happened last night when attempting to print a large helmet model. Printer began making alarm sounds and there was a klipper error that resulted in me having to restart the entire printer.

Printer works when uploading files that are directly in its storage, just not working on Orca.

OddJobEntertainment, 2025-09-29:
Getting same error on my SV08.  I'm able to interact with the printer in every other means remotely, but file transfers aren't happening.

Error uploading to print host:
HTTP 502: <html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx/1.18.0</center>
</body>
</html>

OddJobEntertainment, 2025-09-30:
[image]

OddJobEntertainment, 2025-09-30:
I updated sections of my klipper install and was able to send a file last night.  I'll keep testing, but it seems that it may have been an incompatibility on that end.