OrcaSlicer/OrcaSlicer #10619: When loading a 3D model, the model is not displayed, only a white canvas.

Dmitry-Chernikov, 2025-09-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

SoftFever io.github.softfever.OrcaSlicer 2.0.0-951fc8e master x86_64 orcaslicer3-origin user

### Operating System (OS)

Linux

### OS Version

Kernel: 6.8.0-79-generic x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 5.4.12 tk: GTK 3.24.33     wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy

### Additional system information

Компьютер
******************

Общая информация
-------------------------------

-Компьютер-
Процессор		: AMD Ryzen 9 5950X 16-Core Processor
Память		: 131804 Мб (3440 Мб занято)
Тип машины		: Настольный компьютер
Операционная система		: Linux Mint 21
Пользователь		: dmitry (dmitry)
Дата/время		: Вт 12 авг 2025 14:29:10
-Дисплей-
Разрешение		: 3840x1080 пикселей
Рендер OpenGL		: GeForce GTX 580/PCIe/SSE2
Поставщик X11		: The X.Org Foundation
-Аудио устройства-
Звуковая карта		: HDA-Intel - HDA NVidia
Звуковая карта		: HDA-Intel - HD-Audio Generic
-Устройства ввода-
 Power Button
 Power Button
 Microsoft Wired Keyboard 600
 Microsoft Wired Keyboard 600 Consumer Control
 Microsoft Wired Keyboard 600 System Control
 Logitech Wireless Receiver Mouse
 Asus WMI hotkeys
 Eee PC WMI hotkeys
 HDA NVidia HDMI/DP,pcm:3
 HDA NVidia HDMI/DP,pcm:7
 HD-Audio Generic Front Mic
 HD-Audio Generic Rear Mic
 HDA NVidia HDMI/DP,pcm:8
 HDA NVidia HDMI/DP,pcm:9
 HD-Audio Generic Line
 HD-Audio Generic Line Out Front
 HD-Audio Generic Line Out Surround
 HD-Audio Generic Line Out CLFE
 HD-Audio Generic Front Headphone
-Принтеры (CUPS)-
Canon-LBP710C-PPD		: <i>Default</i>
-SCSI диски-
ATA INTEL SSDSC2KW01

Операционная система
---------------------------------------

-Версия-
Ядро		: Linux 5.15.0-151-generic (x86_64)
Версия		: #161-Ubuntu SMP Tue Jul 22 14:25:40 UTC 2025
Библиотека C		: GNU C Library / (Ubuntu GLIBC 2.35-0ubuntu3.10) 2.35
[... 1418 lines cut ...]

Adventurer 5M

### How to reproduce

Opening a 3D model

### Actual results

[OrcaSlicerLog.txt](https://github.com/user-attachments/files/22110136/OrcaSlicerLog.txt)

https://github.com/user-attachments/assets/2701c955-860b-4aea-92e5-7b12a2f10e99

### Expected results

Opening a 3D model

### Project file & Debug log uploads

[OrcaSlicerLog.txt](https://github.com/user-attachments/files/22110176/OrcaSlicerLog.txt)

### Checklist of files to include

- [x] Log file
- [ ] Project file

DARKFiB3R, 2025-09-03:
Install a newer version of Orca?

Dmitry-Chernikov, 2025-09-03:
> Install a newer version of Orca?

Hello, yes I used apimage V2.3.0 and it was the same, BambuStudio had the same problem, but the update came and it worked, ocaslaiser has a problem. I'll try to install the flatpak package of the new version 2.3.0.

DARKFiB3R, 2025-09-06:
Oh, ok. Your description said you have: SoftFever io.github.softfever.OrcaSlicer 2.0.0-951fc8e master x86_64 orcaslicer3-origin user

Does this happen with all models when you try to open them?

ianalexis, 2026-02-06:
Check current nigthly please

Dmitry-Chernikov, 2026-02-07:
> Check current nigthly please

Now I can’t check, I quit the company that used the equipment.