OrcaSlicer/OrcaSlicer #10633: "Bridge Counterbore Holes" - "sacrificial layer" does not detected all holes, but "Partially bridged" does

ElEsido, 2025-09-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 - alpha

### Operating System (OS)

Windows

### OS Version

Win 10

### Additional system information

CPU: Amd Ryzen 5 3600
Memory: 16GB

### Printer

Bambulab P1S

### How to reproduce

1. Open a part with counterbore holes.
2. Under "Support" disable supports. 
3. Under "Quality - Bridging - Bridge Counterbore Hole" choose "Partially Bridged" and slice. Observe that all counterbore holes are detected and partially bridged. 
4. Now in the same setting choose "Sacrificial layer" and slice. Observe that NOT all counterbore holes are detected and given a sacrificial layer.
5. rotate part, slice again with setting "Sacrificial Layer" - repeat a couple of times. Observe that the number of identified counterbore holes changes. 

### Actual results

Option "Partially Bridged", all holes detected and partially bridged.
[image]

Option "Sacrificial Layer", one counterbore hole is detected and has a sacrificial layer
[image]

Part rotated, option "Sacrificial Layer", two counterbore holes are detected and have a sacrificial layer, two are not detected and the hole is printed in free air. 
[image]

Further rotating results in 0-3 holes being detected, but never all 4. 

### Expected results

When choosing "Sacrificial Layer" all counterbore holes should be detected all the time, just like when choosing "Partially Bridged".

### Project file & Debug log uploads

[CounterboreHoles.zip](https://github.com/user-attachments/files/22166190/CounterboreHoles.zip)

[log.zip](https://github.com/user-attachments/files/22166206/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

discip, 2025-09-05:
@ElEsido
Could you test this #10565?

Link to download:
https://github.com/SoftFever/OrcaSlicer/actions/runs/17304272227

discip, 2025-09-05:
@ElEsido
Could you test this #10565?

Link to download:
https://github.com/SoftFever/OrcaSlicer/actions/runs/17304272227#:~:text=OrcaSlicer%2DLinux%2Dflatpak_PR,OrcaSlicer_Windows_PR10565_portable

discip, 2025-09-05:
Unfortunately the only option left is:
Using support. 😅

ElEsido, 2025-09-05:
:-D
I did that in this model, but am still puzzled by the differential treatment of "partially bridged" and "sacrificial layer". Do you know what goes on in the background? Is there a "counterbore hole detection routine" that differs between the two settings?

Peoplles, 2026-09-04:
Also seeing this issue on 2.4.2.  Was able to partially fix it by rotating the model 45°.

[image]

[image]

discip, 2026-09-21:
@ElEsido
Should be solved.

[image]

<br><br>

Please try the latest dev build:
[a5d3aae](https://github.com/OrcaSlicer/OrcaSlicer/actions/runs/35558172862#:~:text=issues/14748%22-,Artifacts,-Produced%20during%20runtime)

discip, 2026-09-21:
@ElEsido
Should be solved by https://github.com/OrcaSlicer/OrcaSlicer/pull/13956.

[image]

<br><br>

Please try the latest dev build:
[a5d3aae](https://github.com/OrcaSlicer/OrcaSlicer/actions/runs/35558172862#:~:text=issues/14748%22-,Artifacts,-Produced%20during%20runtime)