OrcaSlicer/OrcaSlicer #10642: NFS shares not available

Aaron172-cmd, 2025-09-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-alpha

### Operating System (OS)

Linux

### OS Version

Fedora Workstation 42

### Printer

Creality Ender 3 v2 running Klipper

### How to reproduce

In Orca slicer, when I want to add a file, I click the add icon. Then, in the select menu, I want to change directory to "/mnt/3Dprint/", which is my NFS share, but the slicer doesn't see this as an existing path. This is kind of annoying, because everything 3D printing related is stored on my NAS and not my actual laptop.

### Actual results

[image]

### Expected results

When I open the select file menu, I should be able to choose my NFS share directory to choose files. It's interesting, because in older versions, like 1.8.3 (I think I got the number right) this wasn't a problem.

### Project file & Debug log uploads

Sorry, I don't have any.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hitec-redneck, 2025-09-08:
This is not a bug but a permissions issue if you are using a flatpak. If you use KDE you can use systemsettings to allow the flatpak app to access your nfs mount. If using Gnome you can use an app like Flatseal to configure flatpak permissions

devnoot, 2026-01-03:
I am using the AppImage and seeing the same behavior