OrcaSlicer/OrcaSlicer #10648: Calibration->Flow Rate size is wrong for Voron 0

real-artswan, 2025-09-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-alpha

### Operating System (OS)

Linux, Windows

### OS Version

any

### Additional system information

doesn't matter

### Printer

Voron 0

### How to reproduce

1. Calibration
2. Flow Rate -> Any option 
3. Set of plates appear
4. They don't fit to plate

### Actual results

Calibartion plates don't fit to bed size

[image]

### Expected results

Plates must fit to bed

### Project file & Debug log uploads

[Orca YOLO Flow Calibration.zip](https://github.com/user-attachments/files/22197373/Orca.YOLO.Flow.Calibration.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

MxBrnr, 2025-09-07:
Not really a bug since it was designed to fit on the most popular small size, which is ~180 (Prusa Mini, Bambu Lab A1 Mini, etc.). There are very few printers smaller than that, with the Voron 0 likely being the most popular of such category. It's easy enough to resize the X/Y to fit smaller beds, but I agree it would be nice to have it automatically fit, and would probably be pretty easy to recode the size.

real-artswan, 2025-09-07:
I understand that it's not critical but still would be fine to fix, it's about UX. Why not to take into account bed size and fit?

real-artswan, 2025-09-07:
Now Voron 0 owners have options:
1. Remove some plates and rearrange. Time consuming in preparation and brake the whole idea of plates.
2. Split to 2 beds. Time consuming on preparation and printing longer.
3. Scale XY. You then must uncheck uniform scaling and put 95% in X and Y.
All options are about clicking, tackling, etc... Not convinient. Is not appropriate to the best slicer Orca