OrcaSlicer/OrcaSlicer #10655: Deleting parent printer makes copies unusable

barrelltitor, 2025-09-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Irrelevant

### Printer

Ender 3 Pro

### How to reproduce

1. Create main parent printer
2. Make copies of it
3. Delete the parent printer

### Actual results

The parent printer is deleted alongside all of its configurations existing on the machine, making everything else completely useless

### Expected results

The parent configuration should get merged into the copies

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/22199195/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

It is impossible to export the project file as I have no project, my configurations are gone

I was lucky enough to have an older project with the configurations seemingly saved inside it so now I still have something to go for, but all my history for recent projects seems to be gone too

I ticked the project file box just in case there's some auto closing bot, I hope that is fine.

MxBrnr, 2025-09-08:
Yes, that is how parent/dependencies work. Any "copies" of the parent are considered dependents, and require the parent to continue existing. This is how the feature is designed, so no really a bug.

barrelltitor, 2025-09-08:
At least don't allow parents to be deleted until all children are deleted to not break the configs

barrelltitor, 2026-11-01:
Still happens

LJKraus, 2026-11-01:
> 10\. Still happens

When saving a profile with the current Orca versions, you can choose _Detach from parent_, which makes it independent.