OrcaSlicer/OrcaSlicer #10663: Z-Hop Type Not Always Applied Correctly

staal54a, 2025-09-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-alpha

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

AnkerMake M5

### How to reproduce

1. Open a new project and select one of the AnkerMake M5 profiles
2. Add an object that has several different retraction + travel sequences
3. Set z-hop type to spiral and z-hop height to 0.4mm
4. Slice and check whether all z-hops are of spiral type (may only occur with certain models, see attached project)

### Actual results

Some Z-Hops are just normal/regular and not of the specific type applied (such as spiral or slope). 

### Expected results

I would expect every Z-Hop to be of the type specified.

### Project file & Debug log uploads

[missing_zhop_example.zip](https://github.com/user-attachments/files/22223734/missing_zhop_example.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

This is for sure happening with AnkerMake M5 profiles and it may be happening with other printer profiles. When I first looked into it a while ago (see issue https://github.com/SoftFever/OrcaSlicer/issues/8285) I did notice it with another one of my printer profiles if I rotated the same model so it could be affect several profiles with certain geometry.

There's another question/area this makes me wonder about and it's the Auto setting. I would think Auto should never choose a Normal style Z-Hop as Slope or Spiral are always better (at least in my experience). But I did notice recently on a different printer I had where Auto would cause a lot of stringing on the same model because it chose Normal Z-Hop for some retractions, for some reason.

staal54a, 2025-09-15:
So after doing some more digging, I think one way to force the correct behavior, at least on the Ankermake M5, is to change the `Print sequence` on the plate to `By object` (versus the default value of `By layer`). See the following screenshot where I use the same printer and model as above but simply change the print sequence to `By object`:

[image]

It's a little tough to see but every retraction now correctly uses spiral z-hop.

This is not a solution, however, as it requires changing that print sequence setting which really limits number of models and model geometries that you can print concurrently. But I'm hoping that's enough of a breadcrumb trail that could point at the real underlying issue.

As a note, this bug found the same issue but I think the connection between correct z-hop type and this setting was not made at the time: https://github.com/SoftFever/OrcaSlicer/issues/7854

staal54a, 2025-09-18:
Still seeing the same behavior in 2.3.1-beta

Wegerich, 2026-08-31:
@IguanaBob your problem is different to the original one, I believe it is in fact https://github.com/OrcaSlicer/OrcaSlicer/issues/14568