OrcaSlicer/OrcaSlicer #10708: Filename_format template error: extruded_weight_total "is not a valid variable"

ccatlett1984, 2025-09-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-alpha

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Printer

Bambu X1C

### How to reproduce

1. click "other" tab
2. add `{extruded_weight_total}` to the filename template
3. slice the object, export as gcode.

### Actual results

receive the following error:
```
Processing of the filename_format template failed.
Parsing error at line 1: Not a variable name
{input_filename_base}_{filament_type[0]}_{print_time}_{extruded_weight}.gcode
```

### Expected results

exported filename should contain the amount of filament used.

### Project file & Debug log uploads

can reproduce with any file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ccatlett1984, 2025-09-13:
:(

material used would be a very handy variable in the filename.

Wegerich, 2026-02-25:
This issue appears to be fixed in v2.3.2-beta, though I can't find the pull request in which it was done.