OrcaSlicer/OrcaSlicer #10711: First layer support toolpath still not 100% normal (vs 2.3.0) with 2.3.1 beta

flint723, 2025-09-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 beta

### Operating System (OS)

Windows

### OS Version

Window 11

### Additional system information

14th gen intel 14600k cpu
48gb ddr5 ram
Nvidia 3080ti gpu

### Printer

Qidi Plus4

### How to reproduce

Compare the included project file with other slicers (bambu, Qidi studio) for first layer tree support toolpath.

### Actual results

Support first layer toolpath does not always print outer wall first.  If this is expected I can just close this issue.  But it is different from support toolpath behavior in 2.3.0.

### Expected results

Unsure if this is expected or not. just different from 2.3.0.

### Project file & Debug log uploads

[LS3+Block+++Spark+Plug+Holder.zip](https://github.com/user-attachments/files/22317573/LS3%2BBlock%2B%2B%2BSpark%2BPlug%2BHolder.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

SoftFever, 2025-09-15:
Thank you for reporting this. This behavior with the tree support is unexpected, but so far I’ve been experimenting with it and don’t think it causes any harm. I will consider it a low-priority issue for now.

lanpartyceo, 2025-10-05:
+1 on this. 

It is causing me severe issues and a MASSIVE AMOUNT of cancel and retry on jobs because of completely inconsistent application of support interface first layer. Literally on the SAME Job generation, of which 2 identical (clone) models are sliced and printed, the first support interface printed will print the outer wall first, then print the infill for the support. the next one will print the infill for support first (with NO WALL to adhere to and keep it from curling if it wants).

The expected (and historical) behavior for first layer support interface printing is to print the outer wall first, then the infill. This has ALWAYS been the case. Now it seems, in 2.3.1, support interface will generate in whatever order it feels like, ignoring any settings or modifiers to print outer wall first even when explicitly instructed to do so.

Basically, I can no longer trust ANY job sliced with orcaslicer that requires supports to print the first time. It likely takes 10 or more tries to get past the first layer due to this bug.

chaoscentral, 2025-10-08:
I will add a +1 to this. I am getting FAR more first layer fails with 2.3.1 than 2.3.0 on my Qidi Plus 4. I was able to confirm the issue is printer specific and version specific as changing to another printer solves the issue, or changing back to 2.3.0 also solves the issue. On most of my prints the first layer usually contains no support material or raft material and is a huge issue with getting things to print correctly. 

I'm including an image of the first layer on 2.3.0 vs 2.3.1. 2.3.1 causes that first layer to be printed in the air which makes it literally unusable on most of my prints.

[image]

lanpartyceo, 2025-10-08:
@pm0u Pick any more complex model with overhanging areas like a a miniature or a bust or some sort. That should guarantee you see the issue. also ad multiple copies of the model to the bed for more "chances" of seeing it.

For my part, I started noticing when i was arranging support testing with these models: https://www.printables.com/model/1038477-support-removal-test-no-more-stuck-supports/files

Occasionally it can do it correctly/historically correct, but that often means adding a height range modifier the the print, restricting it to the first layer however this reacts different to different models and some will then print the support order correctly and some will not. **This is NOT guaranteed.**

Here are some examples. In the two mech model images (A and B) we can see the modifier having an effect, but again, not a guarantee. IN the next two for the support structures (C and D) , even though i loaded 6 models, i did not even have to go further than the support printing for the first object than to see the issue on display. the first support base section prints inner/outer, then the VERY NEXT support base section for the object being printed prints outer/inner. 

There is zero consistency to support base print order in OrcaSliver 2.3.1-2.3.2. 

(A)[image]

(B) [image]

(C)[image]

(D)[image]

Reproduce:

1. Download file: https://www.printables.com/model/1038477-support-removal-test-no-more-stuck-supports/files
2. Load 1 or more support .stl files from the download in OrcaSlicer 2.3.1-2.3.2
3. turn of supports and select preferred style.
4. Slice and navigate to the beginning of the first layer.
5. Slide the slider to advance through the proposed motion path.
6. Observe the order the support base sections are printed in. Thew will be printed primarily inner/outer, with a random scattering of outer/inner mixed in.

Expected result:
1. Repeat steps 1-5 above.
2. ALL support base sections should print in outer/inner order.