OrcaSlicer/OrcaSlicer #10718: Printing with external filament not possible on X1C

Duglum, 2025-09-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-beta

### Operating System (OS)

Windows

### OS Version

Windows 10 22H2 10.0.19045

### Printer

Bambu X1C

### How to reproduce

1. Load Model
2. Set to external non-AMS filament
3. Press Slice Plate
4. Press Print Plate
5. Deselect "Enable AMS" as usual
6. Press Send

### Actual results

**X1C on firmware v01.08.02.00: Printer prints, no issues**
 
**X1C on firmware v01.10.00.00:**
* Printer gets to the stage where it heats the nozzle to printing temperature
* Nothing happens anymore
* After about 12-14mins Orca displays an error: `Failed to get AMS mapping table; please select "Resume" to retry. [07FF 8012 181114]`

[image]

### Expected results

Printer prints with external filament on firmware v01.10.00.00, no issues.

### Project file & Debug log uploads

[orca_debug.zip](https://github.com/user-attachments/files/22321785/orca_debug.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I am guessing here that this may be related to changes in how the Bambu firmware handles external filament mapping?  
Their Bambu Studio does not have the option to enable or disable the AMS anymore when sending a print, instead the user has to select the mapping as external in the AMS/filament dropdown. See Screenshots for Details.

This also happened with Orca 2.3.1-alpha and a nightly that was inbetween alpha and beta. I did not try older versions.  
This problem still occurs after a complete factory reset of the printer. The printer is in LAN-Only/Dev mode.

[image]
[image]

tina-junold, 2025-09-15:
Related to #10588 and #9568 ?

The1CheetahAssassin, 2025-09-16:
Can confirm this issue occurs on a P1S with AMS on 1.08.02.00 and 1.09.00.00. Printer is not in dev mode, but project was sliced with orca and exported as GCode 3mf that was uploaded to Bambu Connect. Same steps and issue occur: freezes after heating nozzle. Unplugging the AMS allowed it to print the GCode without issue.